Jalle19/satip-axe #3: sendmmsg(): errno 38: Function not implemented

melias122, 2021-01-14:
I tried latest build yesterday, but  it does not work. From minisatip log prints:
```
sendmmsg(): errno 38: Function not implemented
```
so I cannot get any data from satip sever.  It looks like sendmmsg is available from glibc 2.14 and linux 3.0, which satip-axe should be using according to:
```
~ # /lib/libc.so.6
GNU C Library stable release version 2.14.1, by Roland McGrath et al.
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 4.8.4 20141219 (STMicroelectronics/Linux Base 4.8.4-139).
Compiled on a Linux 3.4.100 system on 2016-03-17.
Available extensions:
crypt add-on version 2.1 by Michael Glad and others
Native POSIX Threads Library by Ulrich Drepper et al
BIND-8.2.3-T5B
libc ABIs: UNIQUE
For bug reporting instructions, please see:
<http://www.gnu.org/software/libc/bugs.html>.
```
Any idea what should be the issue?

Thanks,

Jalle19, 2021-01-15:
Does minisatip crash, or does it just not work? I don't have anything related to `sendmmsg` in by logs.

Jalle19, 2021-01-15:
Actually this might be because I'm using TCP. If you're using tvheadend, can you change the protocol to TCP and try again? This needs fixing in minisatip.

Jalle19, 2021-01-15:
Closing this in favor of https://github.com/catalinii/minisatip/issues/795

closed 2021-01-15 by Jalle19

estimadarocha, 2021-01-31:
Any new build with this mods??

Jalle19, 2021-01-31:
> Any new build with this mods??

I'm working on it

Jalle19, 2021-02-01:
https://github.com/Jalle19/satip-axe/releases/tag/build17 UDP streaming should work again with this build

commit ecc47f8d (2021-01-31): Update minisatip to 191fe62a7a5aaada03ef274511b24238c210693c, should fix sending of UDP packets

Closes #3
commit a9e1ffa0 (2021-01-31): Update minisatip to 191fe62a7a5aaada03ef274511b24238c210693c, should fix sending of UDP packets

Closes #3