Jalle19/satip-axe #7: Tuner seems to require "warm up"

bfayers, 2021-02-21:
My setup is a Digibit R1 with one cable from a Unicable LNB with the following options: `MINISATIP_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040"` connected to TVHeadend

The problem I am having is that if nothing satellite has been streamed in a while attempting to start a stream will see TVHeadend (and the minisatip webui) show Tuner 0 as having about 27% Signal and 0% SNR, it will timeout after the ~10 seconds TVH is set to and move on to Tuner 1 and immediately start streaming at 100% signal and 100% SNR.

The issue persists even if the TVH timeout is lowered to 2 seconds or raised to 20.

It's almost as if the frontends go to sleep...?

bfayers, 2021-02-21:
I found this thread on minisatip here: https://github.com/catalinii/minisatip/issues/313

Which lead me to believe it was an LNB Power issue, I then found [this](https://github.com/perexg/satip-axe/issues/165#issuecomment-727170615) comment by  Jalle19 mentioning the use of `-Z *:0` to keep tuners/LNB alive.

I can confirm this resolves the issue. 👍

closed 2021-02-21 by bfayers

Jalle19, 2021-02-22:
This is precisely the issue 👍