Jalle19/satip-axe #8: Not working

VenimK, 2021-04-17:
Been trying to use the release 16 and 17
When booting the USB stick i get the knight rider effect.

but when I go to the site ip of the digibit i still get the site from the digibit itself
Using TVheadend with 1 cable on input 1 (disecq) Four Sats 13,19,23,28
Any pointers here ....

Jalle19, 2021-04-18:
I think you have to wait for longer before you try accessing the web interface again. You can also try flashing version 15 in case there's something in my builds that prevent them from being flashes on top of stock firmware (something I haven't tried, but I'm assuming it should work).

VenimK, 2021-04-18:
After a few try 's it worked
in sysconfig i disabled the Minisatip8
And changed it 
MINISATIP="yes"
MINISATIP_OPTS="--diseqc-timing *:155-54-15-15-15-0 -7 0:1,0:2,0:3"

'Minisatip Version: 1.0.4-191fe62 -- Build date/time: Jan 31 2021 | 14:50:11'

Jalle19, 2021-04-18:
That is okay, it should work despite the scary messages.

closed 2021-04-18 by Jalle19