Jalle19/satip-axe #9: minisatip "The connection was reset"

j0nnymoe, 2021-05-22:
Hi,

Currently experiencing an issue with minisatip within the latest build. Flashed your firmware over the top of perexg's older release and updated my config file. All seemed to flash great but having issues with minisatip. On inital boot I can get to the webui but shortly after it stops responding and I get "The connection was reset".

Config file below, Thanks :)

```
#
# SATIP AXE firmware configuration
#

CFGVER="15"
HOSTNAME="satip-axe"
KMODULES="drivers/usb/serial/cp210x.ko drivers/usb/serial/pl2303.ko drivers/usb/serial/ftdi_sio.ko"

#
# STV6210 baseband amplifier gain (0dB .. 16dB - 2dB step)
#
STV6210_BAMP="16dB"
#DEMUXTS_OLD="yes"

#
# ttyUSB names
#

TTYUSB1_DEV="2-1:1.0"
TTYUSB1_NAME="ttyUSBupper"
TTYUSB2_DEV="4-1:1.0"
TTYUSB2_NAME="ttyUSBlower"

TTYUSB3_DEV="1-1.1:1.0"         # USB hub port#1 in upper port
TTYUSB3_NAME="ttyUSBupper1"
TTYUSB4_DEV="3-1.2:1.0"         # USB hub port#2 in lower port
TTYUSB4_NAME="ttyUSBlower2"

#
# IPV4 config
#
IPV4="dhcp"
# or
#IPV4="192.168.1.240"
#IPV4_NETMASK="255.255.255.0"
#IPV4_BROADCAST="192.168.1.255"
#IPV4_GATEWAY="192.168.1.1"

#
# ntpd
#
NTPD_OPTS="-p 0.uk.pool.ntp.org"

#
# syslogd
#
#SYSLOGD_OPTS="-R 192.168.1.1:514"

#
# telnetd
#
TELNETD="yes"

#
# dropbear
#
DROPBEAR="yes"

#
# minisatip
#
MINISATIP="yes"
MINISATIP_OPTS="-e 0-1"

#
# inetd (httpd,ftpd)
#
INETD="yes"

#
# crond
#
CROND="yes"

#
# nfsd v3
#
NFSD="yes"
NFSD_HOTPLUG_EXPORT="*" # replace with host or subnet, '*' = world

#
# oscamd
#
OSCAMD="yes"

#
# custom daemon/service/setup executed from sysinit
#
#CUSTOMD="/etc/sysconfig/myboot"
```

Jalle19, 2021-05-22:
See what `logread` says, and see if the device responds to pings when you get "The connection was reset" in your browser. Check if minisatip crashes too (you can see if the PID changes).

j0nnymoe, 2021-05-23:
See the attached the logs via txt (not to spam the issue). I've done `logread` and `ps x` of 2 different configs. I stock with all 4 tuners enabled then the other with just `-e 0-1` applied to minisatip as I only have 2 feeds going to my R1.

Currently I do have my R1 running fine using the stock config, error seems to come when I limit it to 2 tuners only.

[ps.txt](https://github.com/Jalle19/satip-axe/files/6527790/ps.txt)
[stock-ps.txt](https://github.com/Jalle19/satip-axe/files/6527792/stock-ps.txt)
[stock-log.txt](https://github.com/Jalle19/satip-axe/files/6527793/stock-log.txt)
[log.txt](https://github.com/Jalle19/satip-axe/files/6527794/log.txt)

Jalle19, 2021-05-23:
Can't see anything out of the ordinary. Does TV watching work?

j0nnymoe, 2021-05-23:
It does now. It seems for whatever reason the `-e 0-1` is causing minisatip to crash, Had it running fine stock all day.

Jalle19, 2021-05-24:
I doubt that has to do with the satip-axe firmware itself, sounds more like a bug in minisatip. You might wanna open an issue about it at https://github.com/catalinii/minisatip/issues.

closed 2021-05-24 by Jalle19