Jalle19/satip-axe #11: Multistream support?

tipexfm, 2021-08-17:
Hello,

I tried the latest firmware release from this fork today. Unfortunately, it couldn't tune in to any multistream signal. 
I had to roll back to the latest version uploaded by perexg (2018) to get it working again. I'm using Digibit R1.

Did I miss something or do you confirm that multistream reception is no longer working?

Thank you and keep up the good work!

Jalle19, 2021-08-20:
I don't have access to any multistream transponders, so I haven't been able to test. Can you open an issue about it over at https://github.com/catalinii/minisatip/issues ?

Jalle19, 2021-09-07:
Closing in favor of https://github.com/catalinii/minisatip/issues/897

closed 2021-09-07 by Jalle19