Jalle19/satip-axe #12: Please add minisatip build with dvbapi support

SmoothMC, 2021-08-20:
Hey Jalle19,

thanks for this update version of satip-axe it runs really stable on my ggs.box
but the new Version of minisatip in your build doesn't have the dvbaip feature enabled.
can u please add this soon

thanks for support

Jalle19, 2021-08-22:
As far as I know, DVBAPI support is disabled because the device doesn't have enough CPU resources to decrypt channels.

I can create a test firmware for you if you want to try?

SmoothMC, 2021-08-22:
this would be very nice is there a way that you use the latest release of minisatip then?
At the moment i use "minisatip 1.1.43" with your firmware, but i not get it to built it by my self

thx

Jalle19, 2021-08-23:
Sure

Jalle19, 2021-08-28:
I made a test build with DVB-CSA support, see the pull request

commit 6fff88bf (2021-09-24): Build minisatip with DVB-CSA support (closes #12)
closed 2021-09-24 by Jalle19