Jalle19/satip-axe #13: Low signal using diseqC with Digibit R1 alt. firmware

roisiano, 2021-08-21:
Hello:

I did this hardware mount with a Digibit R1:

1) LNB unicable from Astra 19.2 connected to input 0 of Digibit R1
2) DiseqC 2.0 4x1 (this one: https://www.diesl.es/Conmutador-Diseqc-4x1-de-950-2400-MHz-Exterior-Dintel ) connected to input 2 of Digibit R1:
- Position 1: Astra 19.2 (legacy output from this unicable LNB: https://www.amazon.es/gp/product/B084662RQ1/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 )
- Position 2: Hispasat 30 W
- Position 3: Hotbird 13 E
- Position 4: Hotbird 16 E

3) Othe diseqC (same model) 2.0 4x1 connected to input 3 of Digibit R1:
- Position 1: Astra 19.2 (using the other legacy output from this unicable LNB: https://www.amazon.es/gp/product/B084662RQ1/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 )
- Position 2: Hispasat 30 W
- Position 3: Hotbird 13 E
- Position 4: Astra 23.5 

I'm using different dishes for all this sats, LNB twins, etc.
The hardware mount is tested and are OK. I promise  :mrgreen: . I have tested all the "wires" and they work, 

I'm using this alternative software for Digibit R1, based on minisatip:
https://github.com/Jalle19/satip-axe/releases/tag/build18
 
Configuration:
MINISATIP="yes"
MINISATIP_OPTS="-j 0:0-1210,1:1-1420 --dmx-source 0-1:0,2:2,3:3" 

I "order" Digibit to use 2 unicable and assigning the tuners 0 and 1 to input 1. Tuner 2 to input 2 (1st diseqC). Tuner 3 to input 3 (2nd diseqC).
 

 
In TvHeadend:
* Tuner 1: only position 1 enabled. Astra 19.2
* Tuner 2: only position 1 enabled. Astra 19.2
* Tuner 3:
- Position 1: Astra 19.2
- Position 2: Hispasat 30 W
- Position 3: Hotbird 13 E
- Position 4: Eutelsat 16 E
* Tuner 4:
- Position 1: Astra 19.2
- Position 2: Hispasat 30 W
- Position 3: Hotbird 13 E
- Position 4: Astra 23.5 E 

 
It should word OK, shouldn't it?
Well... it's NOT the case.

Astra 19.2 E working OK in all the positions (OK for unicable, OK for 2 diseqCs; with diseqCs the signal and the SNR are lower).
Other sats: the Digibit "tries" to scan the sats, but NO ENOUGH SIGNAL IS FOUND.
TvHeadend "recognises" all the inputs and "tries" to scan, but, for Digibit R1, signal is not enough (for example: 65% signal, 15% SNR for a TP). 
 
I have made a lot of tests, and, when I connect a single wire for each sat to Digibit R1 and I configure it, I can scan channels OK and I can see them (if are "visible"). For every sat. 

But, for a reason that I don't know, it seems that with diseqC the signal of each SAT is low, and I only can see Astra 19.2 channels with diseqC. With other sats, low signal. 

 
OK. But I have 2 aditional hardware to test the diseqCs. 
A very old MVision receiver and an LG TV with satellite input. 

When I test both diseqCs with MVision or with LG TV, asigning the positions, etc., I CAN SEE THE CHANNELS OF ALL THE SATS. 

So...
a) Is the Digibit R1 "very bad" with sat signals?
b) Should I use some aditional config for minisatip to use these diseqCs inputs to obtain more signal and better SNR? 

Thanks in advance and kind regards.

Jalle19, 2021-08-22:
When scanning the weak satellites with tvheadend, can you check the minisatip web interface and check that the correct LNB is used for the tuner in question? In my experience, low signal strength and SNR is typical when something is misconfigured - the values never really drop to zero.

roisiano, 2021-08-22:
> When scanning the weak satellites with tvheadend, can you check the minisatip web interface and check that the correct LNB is used for the tuner in question? In my experience, low signal strength and SNR is typical when something is misconfigured - the values never really drop to zero.

Well... now it's "working" with Astra 19.2 E, Hispasat 30 W, and Hotbird 13 E in each diseqC. 

https://minisatip.org/forum/viewtopic.php?p=4117#p4117

I suppose it is not really a SW problem. Perhaps I'm not good enough with dish orientation, or perhaps the Digibit R1 is not very good to theal with that. 
 
 
Now I have a small problem, and this perhaps is a bad configuration or a SW problem. 

I can see Astra 19.2 E or Hispasat 30 W channels in the 2 diseqC tuners with no problem at all.
But when I try Hotbird channels (position 3 of diseqC in both), TvHeadend "tries" to open the channel but it can't.
The first try with any channel of Hotbird is always bad.
But, when I try other channel of a different TP, it works with no problem.
When I'm using a tuner with Hispasat and I change to a Hotbird channel in the same tuner, there is no problem.

So... it seems that for any reason, the positions 3 (and suppose 4 too) of the diseqCs are not activated with the first "try" of Hotbird.
Perhaps I should change anything in the config?

MINISATIP="yes"
MINISATIP_OPTS="-j 0:0-1210,1:1-1420 --dmx-source 0-1:0,2:2,3:3" 

Should I use more parameters to "say" Digibit or TvHeadend that "activate" positions 3 (and perhaps 4) of both diseqCs "at the first try" as they do (very well) with positions 1 and 2? 

------

By the way... I tried the option:
-Z *:999 

to "activate" all the tuners for ~ 17 minutes after the last use, but, at least for unicable tuners (0 and 1) or 0 input, it does not work at all.
 
More or less, after 30 seconds of inactivity, when I use a Astra 19.2 channel from unicable, it tries to "open" it with the first tuner (0) and always fails. After 10 seconds, TvHeadend tries the second tuner (1), and there is no problem.
When I'm using a channel in a unicable tuner, I can open another from the other unicable tuner without delay. 

In summary... it seems that -Z *:999 option is not working as I thought. 

Perhaps the option -Z *:0 could work, but I prefer that tuners are inactive when nobody is watching no channel for hours. 10-20 minutes are reasonable for me.

Kind regards

Jalle19, 2021-08-23:
> Perhaps the option -Z *:0 could work

Can you try this? I use that myself, hadn't had any issues even after 100 days of uptime.

roisiano, 2021-09-05:
OK. I did some changes.

Now, diseqC 1 connected to input 2; diseqC connected to input 4 of Digibit R1. LNB unicable connected to input 1.

Config:
MINISATIP="yes"
MINISATIP_OPTS="-F -Z *:0 -j 0:0-1210,2:1-1420 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:*3-0,3:*3-0"

Situation:
1) diseqCs are now working OK with 3 sats. **--diseqc 1:*3-0,3:*3-0** seems to be effective to "start" channels from sat in position 3 of diseqCs.

2) -Z *:0 part (for Unicable) seems that is not working at all.
Again, when I try to open the 1st channel from Unicable, alwais "fails" on tuner 0, and after some seconds it tries in tuner 2, and then it starts.

When I quit the channel, the led from the tuner remains ON during 30 seconds, and, after 30 seconds, it goes OFF.
If I try to open another channel (from Unicable) after 1 minute (more than 30 seconds after quitting), again "fails" in tuner 0, etc.
If I try to open another (different TP) channel while other is open, or before 30 seconds after quitting, no problem.

This is not a dramatic problem for me, but it's bad to get EPG from air. 

Kind regards

Jalle19, 2021-09-07:
You should probably open a ticket against minisatip regarding `-Z *:0` not working. All I know is it definitely works for me. For the record the command I use is:

```
minisatip -f -g -j 0:0-1210,1:1-1420,2:2-1680,3:3-2040 -Z *:0 -S 0-2:0,3:3 -U 0-2:3
```

It's a long shot, but you could try moving the `-Z` parameter after the `-j` :shrug: 

I'm going to close this issue since it doesn't seem to be related to this firmware per se. Thanks for the detailed posts though!

closed 2021-09-07 by Jalle19

roisiano, 2021-09-18:
You were right. When I use -Z 1,2:300 AFTER and not before all the -j stuff, it works.

Kind regards

Jalle19, 2021-09-20:
Incredible! 💪 @catalinii is that a bug?

catalinii, 2021-09-20:
-Z 0,2:300 before does not work? Without the use of *

roisiano, 2021-09-20:
I don't live in the place where I have the Digibit R1 and for that reason I can only "change" parameters more or less once each 2 weeks. 
 
Before the last change I had this config:
MINISATIP_OPTS="-F -Z *:0 -j 0:0-1210,2:1-1420 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:*3-0,3:*3-0" 

And with that config -Z option did not work at all.
It did not work (if I remember well) even if I changed different values, as:
MINISATIP_OPTS="-F -Z *:3600 -j 0:0-1210,2:1-1420 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:*3-0,3:*3-0"
or
MINISATIP_OPTS="-F -Z *:300 -j 0:0-1210,2:1-1420 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:*3-0,3:*3-0"

etc. 

I'm not sure if the problem was that "after" -F the software expected other option, or that -Z does not work before -j, but when I used:
MINISATIP_OPTS="-F -j 0:0-1210,2:1-1420 -Z *:3600 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:*3-0,3:*3-0"
or
MINISATIP_OPTS="-F -j 0:0-1210,2:1-1420 -Z *:300 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:*3-0,3:*3-0"
or
MINISATIP_OPTS="-j 0:0-1210,2:1-1420 -Z 0,2:300 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:4-0,3:4-0"
 
-Z parameter worked perfectly well in all the options where -Z was after -j (or not after -F). 
 
Perhaps it was my mistake in the configs.
 
Finally I left the last one:
MINISATIP_OPTS="-j 0:0-1210,2:1-1420 -Z 0,2:300 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:4-0,3:4-0" 

Because I don't want that diseqC adapters (1 and 3) are "on" after using, only unicable (0 and 2) for 5 minutes. Using * it worked for 4 adapters.
 
I did not try -Z 0,2:300 before -j, but if -Z *:300 work after -j and not before, I suppose that -Z 0,2:300 should not work before -j (or after -F).

 
Thanks to both for your help!

catalinii, 2021-09-20:
The reason I ask is because * could be expanded by bash to the name of all files.
 -Z 0,2:300 does not get expanded and if the command line breaks having it BEFORE -j would indicate a bug

roisiano, 2021-09-20:
I'll try to test it when I come back to the place where I have the Digibit R1.

MINISATIP_OPTS="-Z 0,2:300 -j 0:0-1210,2:1-1420 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:4-0,3:4-0"
could be goog to test?

catalinii, 2021-09-21:
Yes

roisiano, 2021-10-02:
I confirm that:
MINISATIP_OPTS="-Z 0,2:300 -j 0:0-1210,2:1-1420 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:4-0,3:4-0" 

WORKS WELL (for -Z).

So, I suppose the problem was 
-F -Z *:3600

-F was before -Z

Kind regards.