Jalle19/satip-axe #25: Autostart Minisatip on Digibit R1

pcr91, 2021-12-20:
Hello! Install the latest version of minisatip. I have the config file configured with the tuners to use only one tuner (I use Universal LNB).

The problem is that minisatip does not start automatically. I have to start it with the command minisatip -F -S 1-3: 0 If I restart the digibit, I have to rerun it for it to work.
What can happen?

Thanks a lot.

Jalle19, 2021-12-21:
Are you using this fork's firmware or perexg's?

pcr91, 2021-12-21:
This firmware. The strange thing is that when I save it to the flash of the Digibit, it also does not turn on the network light at startup. You have to run minisatip manually. I previously had the original firmware prior to this.

Jalle19, 2021-12-21:
Can you post your configuration file here (the one from /etc/sysconfig)?

pcr91, 2021-12-21:
~ # cat /etc/sysconfig/config
#
# SATIP AXE firmware configuration
#

CFGVER="19"
HOSTNAME="satip-axe"
KMODULES="drivers/usb/serial/cp210x.ko drivers/usb/serial/pl2303.ko drivers/usb/serial/ftdi_sio.ko"

#
# STV6210 baseband amplifier gain (0dB .. 16dB - 2dB step)
#
STV6210_BAMP="16dB"
#DEMUXTS_OLD="yes"

#
# ttyUSB names
#

TTYUSB1_DEV="2-1:1.0"
TTYUSB1_NAME="ttyUSBupper"
TTYUSB2_DEV="4-1:1.0"
TTYUSB2_NAME="ttyUSBlower"

TTYUSB3_DEV="1-1.1:1.0"         # USB hub port#1 in upper port
TTYUSB3_NAME="ttyUSBupper1"
TTYUSB4_DEV="3-1.2:1.0"         # USB hub port#2 in lower port
TTYUSB4_NAME="ttyUSBlower2"

#
# IPV4 config
#
#IPV4="dhcp"
# or
IPV4="192.168.1.16"
IPV4_NETMASK="255.255.255.0"
IPV4_BROADCAST="192.168.1.255"
IPV4_GATEWAY="192.168.1.1"

#
# ntpd
#
NTPD_OPTS="-p 3.es.pool.ntp.org"

#
# telnetd
#
TELNETD="NO"

#
# dropbear
#
DROPBEAR="yes"

#
# minisatip
#
MINISATIP="YES"
MINISATIP_OPTS="-S 1-3:0"

#
# minisatip 0.5
#
#MINISATIP5="yes"
#MINISATIP5_OPTS="-L 0:1,0:2,0:3"

#
# inetd (httpd,ftpd)
#
INETD="yes"

#
# crond
#
CROND="yes"

#
# nfsd v3
#
NFSD="yes"
NFSD_HOTPLUG_EXPORT="*" # replace with host or subnet, '*' = world

#
# oscamd
#
OSCAMD="NO"

#
# custom daemon/service/setup executed from sysinit
#
#CUSTOMD="/etc/sysconfig/myboot"
~ #

Jalle19, 2021-12-21:
If you run `logread`, do you see anything related to minisatip?

pcr91, 2021-12-21:
I don't see anything related to minisatip. As I mentioned earlier, the Digibit worked fine with the original firmware. Once I installed this firmware, it would not turn on the network light. Doing some research, start the command to run it manually and it works. The problem is that every time I restart the Digibit I have to do the manual task. I have tried it both by running it on USB and on the FLASH memory of the Digibit and it does the same to me.

pcr91, 2021-12-21:
What could I do to erase all flash memory from the Digibit device and reinstall it again?

Jalle19, 2021-12-21:
You could try https://github.com/perexg/satip-axe/blob/9fcc3cd9897cc81d29ec0d8e7c6cf3fa1754bed5/dist/README#L178

Jalle19, 2022-01-20:
Closing due to inactivity. I hope you figured it out!

closed 2022-01-20 by Jalle19