Jalle19/satip-axe #28: Build 20 problem on Digibit R1

apt1506, 2022-03-24:
I had no issues with build 19, but since I flashed build 20, I can't view any channel on Digibit R1. The picture flashes momentarily while cycling through the tuners. 
I just built with minisatip version 1.1.83 and flashed it, and it works fine again.

estimadarocha, 2022-03-24:
by the way... and nothing related to the issue... how did you updated? usb? the upgrade-fw is not there anymore

apt1506, 2022-03-24:
I used the .flash file and renamed it idl4k.scr. Through the 1st usb port

estimadarocha, 2022-03-24:
@Jalle19 is it possible to return upgrade-fw in the future? i have some of these devices on remote locations that are not really easy to upgrade by usb and i can imagine other's with same problem!

Jalle19, 2022-03-24:
@estimadarocha the script hasn't worked with this fork at any point, and it was too much trouble to port it.

Jalle19, 2022-03-24:
> I just built with minisatip version 1.1.83 and flashed it, and it works fine again.

So you're saying 1.1.83 works but not 1.1.87?

apt1506, 2022-03-24:
correct

apt1506, 2022-03-24:
I built with 1.1.86 as well, and that one doesn't work either

estimadarocha, 2022-03-24:
@Jalle19 i really don't remember but i never ever used it to fetch packages directly...  just used old method... upload the *.fw to device and them upgrade-fw file.fw and that was it

Jalle19, 2022-03-24:
@estimadarocha okay, I didn't know it was possible to flash a local file like that. 75% of the script is about downloading the correct file from GitHub and that's the part I didn't bother porting.

I'll see if I can get the script re-instated.

@apt1506 can you find the newest version that works?

apt1506, 2022-03-24:
@Jalle19 Ok gimme a couple of minutes and I'll let you know ;)

apt1506, 2022-03-24:
@Jalle19  1.1.84 is the latest working minisatip. 85-87 have the same issue.

Jalle19, 2022-03-24:
https://github.com/catalinii/minisatip/releases/tag/v1.1.85 I don't see anything here that could break the build, are you sure it's not something else that's causing your issues?

ncurado, 2022-03-24:
I had the same issue, i.e. a couple of frames and then stopped (tried with VLC and via TVHeadEnd). I didn't try to create my own build.

estimadarocha, 2022-03-24:
so any ideas?

Jalle19, 2022-03-24:
@apt1506 @ncurado I guess I'll make a new build with 1.1.83 instead then 👍

Jalle19, 2022-03-24:
Are you having issues on FTA channels too or just encrypted ones?

ncurado, 2022-03-24:
I only tried with encrypted channels.

Regards,
Nuno

apt1506, 2022-03-24:
> Are you having issues on FTA channels too or just encrypted ones?

FTA channels as well

Jalle19, 2022-03-26:
Making a new build with minisatip 1.1.83 until we have figured out what the problem is.

Jalle19, 2022-03-26:
https://github.com/Jalle19/satip-axe/releases/tag/build21

Jalle19, 2022-07-31:
Bumped into the same issue described here when trying v1.2.7.

Jalle19, 2022-08-01:
Turns out minisatip doesn't just stop after a few frames, it crashes. But since satip-axe automatically restarts it, the end result may look like it just freezes.

commit 676cfdd1 (2022-03-26): Downgrade minisatip to v1.1.83

Fixes #28
closed 2022-08-01 by Jalle19