Jalle19/satip-axe #30: STR / SNR is displayed as 0

psychofaktory, 2022-04-24:
On the status page of the minisatip, SNR and STR are displayed as "0%" in the "Signal" column.
In fact, however, a proper signal is present.
Also on the satfinder page, the values are only indicated with "0%".

How can the actual values be displayed here?

psychofaktory, 2022-04-24:
Seems like it has to do with this Issue:
#issue-1030502349

Jalle19, 2022-04-24:
Hmm, something has broken recently if that's the case. I'm not personally on the latest build (yet) so I can't confirm whether this happens to be as well. I'll check and report back later.

If anyone else can reproduce this, feel free to comment here.

roisiano, 2022-05-04:
Yes. Sometimes it has happened to me too, with different versions of the Jalle19 / minisatip firmware.

On a few occasions I have "fixed" it by changing "Signal scale" in TvHeadend (from 240 to 100, and back again to 240; https://tvheadend.readthedocs.io/en/latest/webui/config_tvadapters/ ), but other times I have only "fixed" it by turning the Digibit R1 off and on again.

It doesn't happen often, but maybe 7-10 times since I got the Digibit R1 (18 months or so). I have no idea about the cause. 

It is not a problem to see the channels, but I like to know the signal.

psychofaktory, 2022-05-05:
Changing the signal scale in TvHeadend didn't make any difference.
But I can confirm that after rebooting the Digibit R1 SNR and STR are shown correctly again.
In TVHeadend, on the status page of the minisatip and the satfinder page as well.

BTW: 
I've read [here](https://discourse.coreelec.org/t/sat-ip-settings-for-digibit-r1-and-tvheadend/5490), that the signal strenght in TVHeadend should be set to 255.
Is this recommendation still correct?

Jalle19, 2022-05-06:
My setup looks like this:

```
Digibit R1 (minisatip) --> minisatip "aggregator" --> tvheadend
Digibit R1 (minisatip) --^
```

The signal strength is displayed in the "aggregator" instance, but not in tvheadend, which leads me to believe that this is a minisatip issue. Signal strength seems to be a hard thing to get right because there are different implementations and different scales.

I also remember reading something somewhere about the AXE devices simply being a bit buggy when it comes to signal strength, i.e. it can work fine but suddenly stop until the device is rebooted. I wouldn't get my hopes up on fixing that.

Jalle19, 2022-07-05:
I don't think there's much I can do about this issue, closing it for now.

closed 2022-07-05 by Jalle19

klada, 2022-07-28:
> I also remember reading something somewhere about the AXE devices simply being a bit buggy when it comes to signal strength, i.e. it can work fine but suddenly stop until the device is rebooted. I wouldn't get my hopes up on fixing that.

@Jalle19 I second that. This is happening to me as well and the issue has also been present on perexg's builds. A reboot of the device fixes this for some days. So I guess this is general issue with these AXE devices.

I am using Unicable BTW and this is where it is happening to me. I never actively noticed this happening when feeding the device with only two legacy multiswitch ports. But I could be wrong there.

Jalle19, 2022-07-30:
I use Unicable exclusively as well

klada, 2022-09-25:
Another quick find: Killing minisatip on the device also fixes this. In my case minisatip is restarted automatically after a single `kill` and the new instance is showing STR/SNR correctly again. No reboot required.

Jalle19, 2022-09-25:
If restarting minisatip helps then it's probably a bug in minisatip and not in the firmware.