Jalle19/satip-axe #34: Digibit does not start anymore

fir3drag0n, 2022-09-16:
Hi,

i just wanted to use your newest build22, but when I push the new unicable setting to the box, it does not start anymore:

I have a unicable setup and I tried to use this settings:

`MINISATIP="yes"
MINISATIP_OPTS="-A 0:1-1210,1:0-1420,2:2-1680,3:3-2040"
`

Any help is appreciated!

ClashC, 2022-09-16:
On my grundig box I could not start Unicable either. Started manually, no stations could be tuned on the client… Due to lack of time I could not test further until now... and I am also very interested in a solution. :)

Jalle19, 2022-09-16:
Try adding `-Z *:0` to the options, and try to tune a channel a second time if it fails the first time.

fir3drag0n, 2022-09-17:
Like this?

MINISATIP_OPTS="-A -Z *:0 0:1-1210,1:0-1420,2:2-1680,3:3-2040

Jalle19, 2022-09-18:
No, before `-A`

fir3drag0n, 2022-09-18:
This does not work either, no channel is tuned

Jalle19, 2022-09-19:
Did it work in previous builds?

Jalle19, 2022-09-19:
Actually you're using the wrong switch, you're supposed to use `-u` or `-j` for Unicable settings, not `-A`. The changes mentioned in the README where for the `-U` switch only (capital U).

I'm sure it'll work once you switch to `-u` or `-j` :+1:

closed 2022-09-19 by Jalle19