Jalle19/satip-axe #39: Problems when try to watch 2 channels of the same transponder

roisiano, 2022-11-13:
Hello:

With version 23:
https://github.com/Jalle19/satip-axe/releases/tag/build23 

It seems that system "becomes crazy" when I try to see 2 channels os the same TP, and I can't see no one of them until I stop one of the 2 channels of the same TP.

Jalle19, 2022-11-14:
The `upgrade-fw` command is only for flash installations - if you're booting from a USB stick you should put the new firmware on the USB and reboot normally (I think, I've never used that mode).

Can you copy the `minisatip` binary from build 22 and upload it to a device running build 23 to see if your issue is resolved then? Then we know whether issue is in minisatip or somewhere else.

Jalle19, 2022-11-14:
Looks like your problem could be related to the `--dmx-source` option. That option was removed in build 22 in favor of another `-A` or `--virtual-diseqc`. You're also only specifying two slots in your Unicable configuration, not sure if that's correct or not.

roisiano, 2022-11-14:
> related to the `--dmx-source` option. That option was removed in build 22 in favor of another `-A` or `--virtual-diseqc`. You're also only specifying two slots in your Unicable configuration, not sure

Yes; but bulid 22 is working OK again with that config. NOT build 23. 

I have 2 diseqCs connected to Digibit R1. For this reason I use only 2 unicable adapters (0 and 2) with the Unicable coaxial cable connected to physical input 1. 

Not sure that the problem is the configuration. I suppose that, perhaps, minisatip ignores this: --dmx-source. When I try to watch channels from different TPs (with Unicable or diseQC), no problem at all. The problem is, with build 22, when I try to see channels from the same TP.

Jalle19, 2022-11-14:
Try copying the minisatip binary from build 22 to build 23

roisiano, 2022-11-15:
> minisatip binary from build 22 to build 23

Sorry, I don't understand exactly what you're proposing me to do. Rename the file " satip-axe-202210230711-23.fw" as " satip-axe-202208090718-22.fw"?

Kind regards

Jalle19, 2022-11-15:
I'll make a custom build for you that you can try.

Jalle19, 2022-11-16:
[satip-axe-202211161729-23-test1.zip](https://github.com/Jalle19/satip-axe/files/10024272/satip-axe-202211161729-23-test1.zip)

@roisiano can you try this build?

roisiano, 2022-11-16:
> [satip-axe-202211161729-23-test1.zip](https://github.com/Jalle19/satip-axe/files/10024272/satip-axe-202211161729-23-test1.zip)
> 
> @roisiano can you try this build?

OK; but when I am at the place where I have the Digibit R1 (maybe within 10 or 20 days).

Jalle19, 2022-11-16:
You can upgrade it remotely using the `upgrade-fw` script

roisiano, 2022-11-19:
> You can upgrade it remotely using the `upgrade-fw` script

No; I use your firmware from USB. I'll check it when I'm in the place where the Digibit is.

Kind regards

estimadarocha, 2022-11-27:
I can confirm @Jalle19 something is really wrong with this version!

Command line: minisatip -f -g -z /etc/sysconfig/minisatip -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 -b 188000 -B 14

Cesbo/astra is getting pids=all from satip-axe... and it gets crazy!

estimadarocha, 2022-11-27:
it seens like the problem in my case is related with pids=all it doesn't work at all!

estimadarocha, 2022-11-27:
add this log: [27/11 16:42:52.701]: tuning to 12360000(1420000) pol: h (2) sr:27500000 fec:  delsys:dvbs2 mod:  rolloff:auto pilot:auto, ts clear=2077, ts pol=2077 sequence_index = 262143
[27/11 16:42:52.701]: axe: isi -1 pls -1 mode -1
[27/11 16:42:52.709]: adapter post_tune: aid 0
[27/11 16:42:52.709]: utils.c:579 allocation_wrapper malloc allocated 4144 bytes at 0x4a9608
[27/11 16:42:52.709]: new filter 0 added for adapter 0, pid 0, flags 20, mask_len 0, master_filter 0
[27/11 16:42:52.709]: utils.c:579 allocation_wrapper malloc allocated 4144 bytes at 0x4aa640
[27/11 16:42:52.710]: new filter 1 added for adapter 0, pid 17, flags 20, mask_len 0, master_filter 1
[27/11 16:42:52.710]: Dumping pids table for adapter 0, number of unknown pids: 0
[27/11 16:42:52.710]: pid 8192, fd 0, packets 0, d/c errs 0/0, flags 2, pmt -1, filter -1, sock -1, sids: 0 -1 -1 -1 -1 -1 -1 -1
[27/11 16:42:52.711]: setting filter on PID 8192 for ADAPTER 0
[27/11 16:42:52.711]: Reply (handle 16) [127.0.0.1:60261] content_len:0, sock 11
[27/11 16:42:52.711]: MSG client << process :
HTTP/1.0 200 OK
Date: Sun, Nov 27 16:42:52 2022 GMT
Server: minisatip/1.2.38
Content-Type: video/mp2t
Connection: close

[27/11 16:42:52.744]: Flushing adapter 0 buffer of 11280 bytes after the tune 33 ms ago
[27/11 16:42:52.747]: Got the new transponder 0026 38, position 188, 35 ms after tune
[27/11 16:42:52.747]: read_dmx send=1, force_send=1, cnt 0 called for adapter 0 -> 14852 out of 188000 bytes read, 2083 ms ago (2311 228)
[27/11 16:42:52.747]: process_dmx start called for adapter 0 -> 14852 out of 188000 bytes read, 2083 ms ago
RECEIVED SIGNAL 11 - SP=49F1D0 IP=49F1D0
Obtained 1 stack frames.
sh: addr2line: not found
0x418f8a :

Jalle19, 2022-11-27:
@estimadarocha can you test the firmware from https://github.com/Jalle19/satip-axe/issues/39#issuecomment-1317393545 ?

estimadarocha, 2022-11-27:
@Jalle19 i used last minisatip releases .47 i think... And no problem... Your build of minisatip is broken

Jalle19, 2022-11-27:
Did you test the build I linked?

estimadarocha, 2022-11-27:
No... Before you asked it i already testes with own ones of minisatip GitHub. Do you want me to test yours?

Jalle19, 2022-11-27:
It would be helpful too, yes. Good to know that 1.2.47 works though, I guess I'll make a new build using that version. Now that I think about it I have had some strange issues with 1.2.39 as well, could be related to this.

Jalle19, 2022-11-27:
Can you both try https://github.com/Jalle19/satip-axe/releases/tag/build24-rc1 and report whether or not it works?

estimadarocha, 2022-11-27:
In 30 mins

estimadarocha, 2022-11-27:
sorry @Jalle19 something is really wrong

[image]

Jalle19, 2022-11-28:
There's nothing new in this build except minisatip v1.2.47. Didn't you say that version worked when you tested earlier?

Jalle19, 2022-11-28:
The upstream binaries are compiled with PMT=0, maybe that's the culprit.

Jalle19, 2022-11-28:
I'd rather revert to the last working version from build 22.

estimadarocha, 2022-11-28:
@Jalle19 can you make a build with PMT=0?

Jalle19, 2022-11-28:
I can't seem to compile it with that option:

```
dvbapi.c: In function 'dvbapi_add_pmt':
dvbapi.c:823:17: error: 'adapter' has no member named 'transponder_id'
     k->tsid = ad->transponder_id;
                 ^
make[2]: *** [../build/dvbapi.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory `/build/apps/minisatip/src'
make[1]: *** [minisatip] Error 2
make[1]: Leaving directory `/build/apps/minisatip'
make: *** [apps/minisatip/minisatip] Error 2
make: *** [Makefile:102: docker-clean-release] Error 2
```

estimadarocha, 2022-11-28:
Any idea about the cause?

Jalle19, 2022-12-18:
```
RECEIVED SIGNAL 11 - SP=4A3120 IP=4A3120
Obtained 1 stack frames.
sh: addr2line: not found
0x4190f6 :
```
Would be helpful to get this working on these builds so we could see where it crashes

estimadarocha, 2022-12-18:
I am fixing everything so @catalinii can have full acess to my system. By the way is there a way to have some linda of vpn packages on this machine? Openvpn or wireguard since its a low power CPU?

Jalle19, 2022-12-18:
```
~ # addr2line -a -e /sbin/minisatip
0x418f8a
0x00418f8a
/build/apps/minisatip/src/utils.c:880
0x4190f6
0x004190f6
/build/apps/minisatip/src/utils.c:963
```

Could be that my build is not identical though so take these with a grain of salt

Jalle19, 2022-12-18:
@estimadarocha https://github.com/Jalle19/satip-axe/releases/tag/build42-rc2 can you try to reproduce any crashes you've been experiencing lately with this build? It has the `addr2line` utility available so it should actually show where the crash happens.

The build contains minisatip 1.2.12, if you can't get it to crash on that version let me know and I'll make you one with the latest release (compiled _without_ PMT=0, because I think that's what making it crash).

estimadarocha, 2022-12-18:
Yes i can. But i believe they wont happen with that version

catalinii, 2022-12-18:
addr2line in base image will speed the troubleshooting. @estimadarocha can you upgrade to that image?

Jalle19, 2022-12-19:
@roisiano can you reproduce the issue with https://github.com/Jalle19/satip-axe/releases/tag/build24-rc3 and a copy of minisatip from build 23?

estimadarocha, 2022-12-19:
@Jalle19 testing [Build 24 (rc3)](https://github.com/Jalle19/satip-axe/releases/tag/build24-rc3)

estimadarocha, 2022-12-19:
> @estimadarocha https://github.com/Jalle19/satip-axe/releases/tag/build42-rc2 can you try to reproduce any crashes you've been experiencing lately with this build? It has the `addr2line` utility [...]

no... with 1.2.12 it works!

Jalle19, 2022-12-20:
[satip-axe-202212201649-24-rc3-test1.zip](https://github.com/Jalle19/satip-axe/files/10270453/satip-axe-202212201649-24-rc3-test1.zip)

@estimadarocha can you try this build? It has minisatip v1.2.49.

roisiano, 2022-12-21:
> @roisiano can you reproduce the issue with https://github.com/Jalle19/satip-axe/releases/tag/build24-rc3 and a copy of minisatip from build 23?

Sorry, I won't be at the place where the Digibit R1 is for a few weeks. If I can, I'll try when I'm there.
Could you tell me how to see and upload the logs?

Kind regards

Jalle19, 2022-12-31:
@roisiano @estimadarocha have you had a chance to tets the build I linked here (24-rc3-test1)?

estimadarocha, 2023-01-04:
@Jalle19 will test today!

catalinii, 2023-01-17:
The current master should work now on axe.

Jalle19, 2023-01-17:
I'll make a new build then 👍

estimadarocha, 2023-01-17:
@Jalle19 @catalinii thanks

roisiano, 2023-01-22:
Hello:

Finally, I could test the new releases with Digibit R1.

**Confirmed! 24 (rc3)
https://github.com/Jalle19/satip-axe/releases/tag/build24-rc3 

Seems to work perfectly for me with different channels of the same transponder. I tested until 5 channels from same TP, and 2+3 from 2 different TPs.**

All OK.

Thanks again and kind regards

roisiano, 2023-01-24:
> @roisiano @estimadarocha have you had a chance to tets the build I linked here (24-rc3-test1)?

https://github.com/Jalle19/satip-axe/releases/tag/build24-rc3 

Works OK with several channels of the same TP.

Kind regards

Jalle19, 2023-03-24:
https://github.com/Jalle19/satip-axe/releases/tag/build24

PR #43 (2022-11-27): Bump minisatip to v1.2.47
Should fix #39
commit 7d1d4825 (2022-11-27): Bump minisatip to v1.2.47

Should fix #39