Jalle19/satip-axe #42: Additional packages not loaded with current satip-axe version

meijkl, 2022-11-21:
I'm using some additional packages like Midnight Commander (see: https://github.com/perexg/satip-axe/issues/150) but just saw that these are no longer loaded with your current version.

Any idea why it doesn't work? 

Klaus

P.S.: And by the way - MANY THANKS FOR KEEPING THE SATIP-AXE SOFTWARE ALIVE  :-)))

Jalle19, 2022-11-22:
I'll make a new release soon, but I'm reluctant to add any additional software

Jalle19, 2022-11-27:
Can you try https://github.com/Jalle19/satip-axe/releases/tag/build24-rc1 ?

Jalle19, 2022-11-27:
Sorry, commented on the wrong issue. But feel free to try the release candidate nevertheless!

closed 2022-11-27 by Jalle19

meijkl, 2022-11-28:
Thank you!
Just read the test results by estimadarocha, it seems that his problem isn't solved but I'll give it a try today!

Jalle19, 2022-11-28:
Yeah I'm thinking of just reverting minisatip to the version used in build 22.