Jalle19/satip-axe #44: NTPD time sync is not working

rajkosto, 2022-12-08:
`/usr/sbin/ntpd -n -I eth0 -p 0.pool.ntp.org` is in the ps aux list
```
PING 0.pool.ntp.org (109.245.222.66): 56 data bytes
64 bytes from 109.245.222.66: seq=0 ttl=55 time=16.458 ms
```
yet
```
~ # date
Thu Jan  1 12:03:48 UTC 2015
```

if i just run `/usr/sbin/ntpd -n -I eth0 -p 0.pool.ntp.org`` manually in the shell it syncs the clock
```
~ # /usr/sbin/ntpd -n -I eth0 -p 0.pool.ntp.org
ntpd: setting time to 2022-12-08 00:55:15.381171 (offset +250347057.295011s)

~ # date
Thu Dec  8 00:55:27 UTC 2022

```

Jalle19, 2022-12-11:
I'm not quite sure what the right thing to do here is. Should minisatip fail to start if no NTP server can be reached? That seems unreasonable.

rajkosto, 2022-12-11:
Just wondering why ntpd doesnt do anything until i manually run it myself
minisatip uptime is a minor thing, it would probably not even report that wrong uptime if ntpd synced at startup before it launched...

Jalle19, 2022-12-12:
For me ntpd always syncs the time on startup, it just happens to do it a little bit too late (after minisatip has started).

Do you use static network configuration or DHCP?

Jalle19, 2022-12-12:
If you specify an NTP servwe by IP address instead of letting it default to 0.pool.ntp.org, does it then eventually update the time correctly? You can override the ntpd command line from /etc/sysconfig/config.

Just want to rule out any DNS issues here.

rajkosto, 2022-12-12:
```
~ # ps | grep ntpd
  720 root       0:00 {ntpd} /bin/sh /etc/init.d/ntpd
  989 root       0:00 /usr/sbin/ntpd -n -I eth0 -p 216.229.4.66

~ # date
Mon Dec 12 23:55:11 UTC 2022
```

this works, restarted it twice
so i guess /etc/resolv.conf is empty because no dhcp yet when that ntpd process is started and it gets stuck on dns resolution ?
minisatip still says this though
`Current Time: Mon Dec 12 23:56:10 2022 -- Start Time: Thu Jan 1 12:00:32 2015 -- Running Time: 2902d 11:55:38`

Jalle19, 2022-12-13:
The ntpd startup script waits until /etc/resolv.conf exists before starting ntpd. If the file is empty that means the DHCP server responded but didn't include a list of DNS servers.

I would suggest you get your DHCP/DNS setup working perfectly first, but like you said this won't help with minisatip uptime. The init system used in this project is very very basic and attempting to add more logic is no fun really.

Jalle19, 2022-12-13:
I think I found a solution, works locally at least:

```
Jan  1 12:00:02 (none) daemon.err udhcpc[718]: sending discover
Jan  1 12:00:05 (none) daemon.err udhcpc[718]: sending discover
Jan  1 12:00:08 (none) daemon.err udhcpc[718]: sending discover
Jan  1 12:00:08 (none) daemon.err udhcpc[718]: sending select for 10.110.4.12
Jan  1 12:00:08 (none) daemon.err udhcpc[718]: lease of 10.110.4.12 obtained, lease time 3600
Jan  1 12:00:08 (none) local0.notice root: DHCP bound ('10.110.4.12' '' 'netmask 255.255.0.0' | '10.110.5.1' '10.110.5.100') replacing ''
Dec 13 22:01:20 (none) local0.notice root: ntpd time synchronized, signaling minisatip startup
Dec 13 22:01:20 (none) user.notice minisatip: Setting jess adapter 0 slot 4 freq 984 pin -1
Dec 13 22:01:20 (none) user.notice minisatip: Setting jess adapter 1 slot 5 freq 1020 pin -1
Dec 13 22:01:20 (none) user.notice minisatip: Setting jess adapter 2 slot 6 freq 1056 pin -1
Dec 13 22:01:20 (none) user.notice minisatip: Setting jess adapter 3 slot 7 freq 1092 pin -1
Dec 13 22:01:20 (none) user.notice minisatip: Setting master adapter 0 for adapter 0
Dec 13 22:01:20 (none) user.notice minisatip: Setting master adapter 3 for adapter 3
Dec 13 22:01:20 (none) user.notice minisatip: Setting source 0 (src=1) to adapter 0 position 0
Dec 13 22:01:20 (none) user.notice minisatip: Setting source 0 (src=1) to adapter 1 position 0
Dec 13 22:01:20 (none) user.notice minisatip: Setting source 0 (src=1) to adapter 2 position 0
Dec 13 22:01:20 (none) user.notice minisatip: Setting source 1 (src=2) to adapter 3 position 1
Dec 13 22:01:20 (none) daemon.notice minisatip[835]: minisatip version 1.2.12, compiled in Dec 11 2022 20:39:58, with s2api version: 0501
```

minisatip will wait for 10 seconds for synchronized time, after which it gives up and starts anyway.

rajkosto, 2022-12-15:
Build so i can try it out with my (borked?) dhcp server ?

rajkosto, 2022-12-16:
ive tried the build from the pull request, both cases work (when the ntp sync fails and it boots up with 2015, and when i use IP for ntp server and it succeeds, and the start time is correct)

Jalle19, 2022-12-16:
Great!

estimadarocha, 2022-12-16:
@Jalle19 i saw some time ago the same issue on @perexg git https://github.com/perexg/satip-axe/issues/109

Jalle19, 2022-12-16:
Yeah, it's an old issue.

rajkosto, 2022-12-17:
ive rebooted it a few times and consistently, using a IP for the ntp server makes it work, using a dns name makes it not work, even though ive put a resolv.conf in /etc/sysconfig now (in case it needs to be there before dhcp replaces it)

Jalle19, 2022-12-18:
Okay. Using 0.pool.ntp.org works for me so I'm not sure what the problem could be in your case. Are you okay with using an IP address for the NTP server? I would like to close this ticket as part of the next build.

rajkosto, 2022-12-18:
Yeah it's fine the minisatip having correct startup time is a good fix

commit 4d02da8c (2022-12-13): Rewrite ntpd startup script, make minisatip wait for time before starting

10 second grace period after which minisatip will start anyway. Fixes a longstanding issue with incorrect uptime counter in minisatip.

Fixes #44
closed 2022-12-18 by Jalle19