Jalle19/satip-axe #45: "Disabled Tuner" Grundig GSS400

rubenkr, 2022-12-09:
Hi, just want to thank you very much for having this repo updated! At the same time I feel guilty and stupid for not getting it running on my GSS400. I ran the following steps to flash the latest RC1 Release as follows:

`/mnt/data # upgrade-fw /mnt/data/satip-axe-202211272007-24-rc1.fw`

I have 4 LNB inputs of which I am using only one for testing purposes. The device boots up and I can login into ssh with the following logs. I am planning to input the GSS400 into TVhead which is already recognized with a variable in the docker command -e RUN_OPTS="--satip_xml http://10.0.0.254:8080/desc.xml".

**Problem**
No tuners are active, logread shows that no devices could be initialized.

**logread**
`~ # logread
Jan  1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.26.2
Jan  1 12:00:00 (none) local0.notice root: AXE drivers load start
Jan  1 12:00:00 (none) local0.notice root: I2C Demod #2(D2) failed
Jan  1 12:00:01 (none) local0.notice root: AXE drivers load finish
Jan  1 12:00:01 (none) daemon.info init: starting pid 719, tty '': '/etc/init.d/network'
Jan  1 12:00:01 (none) daemon.info init: starting pid 720, tty '': '/etc/init.d/ntpd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 721, tty '': '/etc/init.d/nfsd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 722, tty '': '/etc/init.d/satip'
Jan  1 12:00:01 (none) daemon.info init: starting pid 723, tty '': '/etc/init.d/telnetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 724, tty '': '/etc/init.d/dropbear'
Jan  1 12:00:01 (none) daemon.info init: starting pid 725, tty '': '/etc/init.d/inetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 726, tty '': '/etc/init.d/crond'
Jan  1 12:00:01 (none) daemon.info init: starting pid 728, tty '': '/etc/init.d/oscamd'
Jan  1 12:00:01 (none) daemon.err udhcpc[719]: started, v1.26.2
Jan  1 12:00:01 (none) daemon.info init: starting pid 730, tty '': '/etc/init.d/customd'
Jan  1 12:00:01 (none) local0.notice root: DHCP deconfig
Jan  1 12:00:02 (none) authpriv.warn dropbear[724]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan  1 12:00:02 (none) authpriv.warn dropbear[724]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan  1 12:00:02 (none) authpriv.warn dropbear[724]: Failed loading /etc/dropbear/dropbear_ed25519_host_key
Jan  1 12:00:02 (none) authpriv.info dropbear[724]: Not backgrounding
Jan  1 12:00:02 (none) cron.info crond[726]: crond (busybox 1.26.2) started, log level 8
Jan  1 12:00:02 (none) local0.notice root: Setting STV6210 gain to 8
Jan  1 12:00:02 (none) daemon.warn mountd[721]: Kernel does not have pseudo root support.
Jan  1 12:00:02 (none) daemon.warn mountd[721]: NFS v4 mounts will be disabled unless fsid=0
Jan  1 12:00:02 (none) daemon.warn mountd[721]: is specfied in /etc/exports file.
Jan  1 12:00:02 (none) daemon.notice mountd[721]: Version 1.3.3 starting
Jan  1 12:00:02 (none) daemon.err udhcpc[719]: sending discover
Jan  1 12:00:05 (none) daemon.err udhcpc[719]: sending discover
Jan  1 12:00:08 (none) daemon.err udhcpc[719]: sending discover
Jan  1 12:00:11 (none) local0.notice root: DHCP leasefail
Jan  1 12:00:31 (none) daemon.err udhcpc[719]: sending discover
Jan  1 12:00:31 (none) daemon.err udhcpc[719]: sending select for 10.0.0.254
Jan  1 12:00:31 (none) daemon.err udhcpc[719]: lease of 10.0.0.254 obtained, lease time 86400
Jan  1 12:00:31 (none) local0.notice root: DHCP bound ('10.0.0.254' 'broadcast 10.0.0.255' 'netmask 255.255.255.0' | '10.0.0.1' '10.0.0.1') replacing ''
Jan  1 12:00:31 (none) daemon.notice minisatip[987]: minisatip version 1.2.38, compiled in Oct 23 2022 07:06:09, with s2api version: 0501
Jan  1 12:01:23 (none) authpriv.info dropbear[997]: Child connection from 10.0.0.52:52614
Jan  1 12:01:25 (none) authpriv.notice dropbear[997]: Password auth succeeded for 'root' from 10.0.0.52:52614
Jan  1 12:04:31 (none) daemon.notice minisatip[987]: Initializing with 0 devices
`

**~ # vi /etc/sysconfig/config**
```
#
# minisatip
#
MINISATIP="yes"
MINISATIP_OPTS=""
```

**Screenshots**
[image]
[image]

Thank so much for any help to get one tuner up and running!

rubenkr, 2022-12-09:
After digging deeper into the manuals and comments [(I](https://github.com/Jalle19/satip-axe/issues/7#issuecomment-782888997) found out that I have to set MINISATIP_OPTS as follows:
`MINISATIP_OPTS="-u 0:0-1210,1:1-1420,2:2-1680,3:3-2040"
`

With this I get the 4 tuners initialized. However despite there is a coax cable in input 4 this is not recognized by the device. See screenshots below:
[image]
[image]

rubenkr, 2022-12-09:
I need to make an additional post as by a scheduled OFF duration of 30minutes the device now shows initialized four adapters and displays them now as enabled. YEY! 
[image]
[image]

However as adapter0 has the coax cable attached it still shows no signal which is awkward. The same cable shows DVB-S channels on my TV internal tuner and is confirmed to be working. Any ideas what this remaining issue could be?

**logread**
`Jan  1 12:00:31 (none) user.notice minisatip: Set default timeout to 0
Jan  1 12:00:31 (none) user.notice minisatip: Setting unicable adapter 0 slot 1 freq 1420 pin -1
Jan  1 12:00:31 (none) user.notice minisatip: Setting unicable adapter 1 slot 0 freq 1210 pin -1
Jan  1 12:00:31 (none) user.notice minisatip: Setting unicable adapter 2 slot 2 freq 1680 pin -1
Jan  1 12:00:31 (none) user.notice minisatip: Setting unicable adapter 3 slot 3 freq 2040 pin -1
Jan  1 12:00:31 (none) daemon.notice minisatip[986]: minisatip version 1.2.38, compiled in Oct 23 2022 07:06:09, with s2api version: 0501
Jan  1 12:00:31 (none) daemon.notice minisatip[986]: Initializing with 4 devices
Jan  1 12:01:23 (none) authpriv.info dropbear[996]: Child connection from 10.0.0.52:53588
Jan  1 12:01:26 (none) authpriv.notice dropbear[996]: Password auth succeeded for 'root' from 10.0.0.52:53588`

**Command line:**
` minisatip -f -g -z /etc/sysconfig/minisatip -Z *:0 -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040`

**dmesg**
```
[fta_fe] FE Initialization done, res 0
[dmx-ts] ... init 0
USB Serial support registered for cp210x
usbcore: registered new interface driver cp210x
cp210x: v0.09:Silicon Labs CP210x RS232 serial adaptor driver
USB Serial support registered for pl2303
usbcore: registered new interface driver pl2303
pl2303: Prolific PL2303 USB to serial adaptor driver
USB Serial support registered for FTDI USB Serial Device
usbcore: registered new interface driver ftdi_sio
ftdi_sio: v1.5.0:USB FTDI Serial Converters Driver
eth0: device MAC address 00:22:b7:f0:09:02
STMMAC MII Bus: Can't get IRQ 25 (PHY)
 No MAC Management Counters available
PHY: 1:01 - Link is Up - 1000/Full
PHY: 1:01 - Link is Down
eth0: no IPv6 routers present
PHY: 1:01 - Link is Up - 1000/Full
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
```

Jalle19, 2022-12-10:
What type of LNB do you have?

rubenkr, 2022-12-10:
Hi thanks for asking. I have a Quad-LNB with four coac cables. I am only using one as testing purpose. Still I am digging around and found that I am stuck on version minisatip version 1.2.38. upgrade-fw /mnt/data/satip-axe-202211272007-24-rc1.fw does not bring me to the latest version. So maybe it is also related to this?

rubenkr, 2022-12-10:
OK let me finish this right off as my stupidity could not be larger. Lessons learnt: Do flash anything on internal device when you do not know what you are doing. I repaired the bricked device as follows and inspired by this awesome help from [perexg](https://github.com/perexg/satip-axe/issues/121#issuecomment-424235434).

1. Use USB TTL adapter according this [post](https://github.com/perexg/satip-axe/issues/121#issuecomment-424235434)
2. On Linux PC connect to box with sudo screen -L /dev/ttyUSB0 115200,cs8 (remark: be fast to press ENTER in under a second after device power button is pressed, so have the usb ttl already plugged in and screen command executed)
3. Prepare USB thumdrive with fdisk delete all partitions and format as vfat (e.g. mkfs.fat /dev/sdb) as [here ](https://github.com/perexg/satip-axe/issues/121#issuecomment-425799853)
4. insert usb into upper port of the box
5. inside the screen session execute run updatefw (it will read and write the file wait until everything is finished
6. leave the usb stick plugged in the upper port and power cycle box
7. leds will flash in night rider mode wait until LAN led is blinking
8. power off device
9. follow steps marked [here ](https://github.com/perexg/satip-axe/blob/master/dist/README) on idl4k.cfgreset

Only now I again have access to stock firmware. Thanks for everyone documenting and sorry for this rollercoaster one :D

closed 2022-12-10 by rubenkr