Jalle19/satip-axe #46: Copy minisatip configuration from stock firmware on Grund GSSBOX

rubenkr, 2022-12-10:
Hi, I try to get minisatip working with following **setup**:

- Quad LNB
- One coax cable attached to adapter #0     
- Device: Grundig GSS400 SAT >IP latest firmware from [here ](https://telestar.de/produkt/digibit-r1/)

**Problem**
- On Stock FW Version V1.25.0.157 DVB-S scan is working (see screens below)
- On latest minisatip 1.2.47~436a845 Command line: minisatip -f -g -z /etc/sysconfig/minisatip -Z *:0 -W 0 it is not working

How can I simply copy the stock firmwares settings into command line code for minisatip? Sorry for the beginner question but I could not find any solution in the git repos. THank you very much !

**Screens**
[image]

[image]

[image]

Jalle19, 2022-12-11:
As far as I can tell you shouldn't need any specific options. I would recommend installing build 22 instead, there have been many reported issues with builds 23 and 24-rc1.

Since your question is not really related to the firmware but minisatip itself I think you should ask on minisatip.org instead, or the minisatip issue tracker.

rubenkr, 2022-12-11:
Confirmed. This is a too general question no issue.

closed 2022-12-11 by rubenkr