Jalle19/satip-axe #47: Problems when I try to watch 3 channels of the same transponder (24-rc1)

roisiano, 2022-12-10:
Hello:

I have tried right now version 24-rc1: https://github.com/Jalle19/satip-axe/releases/tag/build24-rc1

Some weeks ago I had a problem with version 23. When I tried to use 2 channels from same Transponder (TP), I could NOT with Digibit R1 ( https://github.com/Jalle19/satip-axe/issues/39 ). 

Well... now with 24-rc1 I can see 2 channels for the same TP (I tried with "unicable TPs"), but when I try to start the 3rd channel of the same TP, I can´t see no one of the 3 channels.
It's not very usual that I see 3 channels from same TP, but some times it happens. 

My config still is:
MINISATIP="yes"
MINISATIP_OPTS="-j 0:0-1210,2:1-1420 -Z 0,2:600 --dmx-source 0:0,1:1,2:0,3:3 --diseqc 1:4-0,3:4-0" 

I returned to version 22. With this version (22) I have tried 4 channels of the same TP righ now without any issues.
I will keep using version 22 for the moment.

Kind regards

Jalle19, 2022-12-11:
@catalinii there's definitely something going on with the latest minisatip builds 🤷 Not sure when exactly it broke but seems like some time after 1.2.22.

Jalle19, 2022-12-18:
Seems to be essentially the same bug as reported in https://github.com/Jalle19/satip-axe/issues/47

Jalle19, 2023-03-24:
https://github.com/Jalle19/satip-axe/releases/tag/build24