Jalle19/satip-axe #51: SSH authorized_keys not working?

rajkosto, 2022-12-18:
```
~ # ls -lan /etc/sysconfig/authorized_keys
-rw-------    1 0        0              399 Jan  1  2015 /etc/sysconfig/authorized_keys
~ # ls -lan ~/.ssh/authorized_keys
-rw-------    1 0        0              399 Jan  1  2015 /root/.ssh/authorized_keys
```

put the same key i use on all my other ssh servers there, when I try to log in with putty i get "Server refused our key" as if its not there

Jalle19, 2022-12-19:
Works just fine here :shrug: 

```
negge@palinski:~$ ssh-copy-id -i ~/.ssh/id_rsa.pub root@10.110.4.11
/usr/bin/ssh-copy-id: INFO: Source of key(s) to be installed: "/home/negge/.ssh/id_rsa.pub"
/usr/bin/ssh-copy-id: INFO: attempting to log in with the new key(s), to filter out any that are already installed
/usr/bin/ssh-copy-id: INFO: 1 key(s) remain to be installed -- if you are prompted now it is to install the new keys
root@10.110.4.11's password: 

Number of key(s) added: 1

Now try logging into the machine, with:   "ssh 'root@10.110.4.11'"
and check to make sure that only the key(s) you wanted were added.

negge@palinski:~$ ssh root@10.110.4.11

Welcome to SATIP-AXE server 202210230708-23

~ # cat .ssh/authorized_keys 
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQCWf2MmpHweXC<SNIP> negge@palinski
~ #
```

Jalle19, 2022-12-19:
Hmm, this seems like it's a regression. It works on build 23 but not 24-rc2

Jalle19, 2022-12-19:
The issue seems to be: `authpriv.info dropbear[2030]: /root must be owned by user or root, and not writable by others`

closed 2022-12-19 by Jalle19

Jalle19, 2022-12-19:
Fixed in 24-rc3 (https://github.com/Jalle19/satip-axe/releases/tag/build24-rc3)

PR #52 (2022-12-19): Don't preserve ownership when copyin in fs-add
We don't want the local copy's file permissions to be reflected in the target firmware filesystem. Should fix #51.
commit 59705f6c (2022-12-19): Don't preserve ownership when copyin in fs-add

We don't want the local copy's file permissions to be reflected in the target firmware filesystem. Should fix #51.