Jalle19/satip-axe #53: usage of internal diseqc-switch

fkzle, 2022-12-23:
parameter "--free-inputs" of perexg. 

My setup looks like this:
inputs 1-3: external diseqc-switch with Astra 19.2° and Astra 28.5°
Input 4: directly connected to LNB for Hotbird 13°. 

I used the "free-inputs" parameter to configure this behaviour. 

my client (VDR) was able to receive all positions. 
After update to your branch, the paramater is not working anymore. 
The "--virtual-diseqc" apparently does something else.  I am not receiving the 13° satellite anymore. 
I can sort out that also with a properly crafted http URL (with all sat parameters correctly set), it is not possible to receive this position. 

How can i achieve this functionality with the stock minisatip? How can i leverage the power of the internal diseqc-switch of the digibit R1?

Jalle19, 2022-12-24:
What values did you use for --virtual-diseqc before?

fkzle, 2022-12-24:
I used the following:
0:0:0,1:0:1,0:1:0,1:1:1,0:2:0,1:2:1,2:3:0

Jalle19, 2022-12-24:
So the only thing not working anymore is Hotbird on input 4?

fkzle, 2022-12-24:
Exactly.
Inputs 1-3 with both positions are working.

fkzle, 2022-12-25:
In the screenshot, i tried to watch "kabelio info" on hotbird (11526MHz H, PIDs: 4531 and 4532). 
[image]

It looks like the mapping is affecting the tuner, not the input of the switch. 
(Tuner 3 still uses LNB1)

Jalle19, 2022-12-25:
I'll get back to you after Christmas. I believe you need to use the -S switch as well to tell minisatip that src=3 should use LNB4.

fkzle, 2022-12-25:
I am not completely sure how this should help. Every input has its own cable to the (external) diseqc-switch, none of them are linked together.

Jalle19, 2022-12-26:
> This feature is used by FBC receivers and AXE to specify the source input of the adapter

fkzle, 2023-01-03:
I tried your suggestion, but you are correct, this is rather unintuitive (and sadly also badly documented).

I tried "-S 0:0,1:1,2:2,3:3" and expecting that a certain tuner should use its corresponding input (tuner 2 using input 2). This seems not to be the case. 

In addition, i got a lot of  "connect failed" on my satip-client. (though this may be unrelated)

[image]

Jalle19, 2023-01-04:
Personally I have one Unicable LNB connected to input 1 and another Unicable LNB connected to input 4, and I have it configured so that the first three tuners use input 1 while the fourth tuner uses input 4. This is done with `-S 0-2:0,3:3 -A 0:0:0,0:1:0,0:2:0,1:3:1`.

Do you have physical cables attached to each of your inputs, or only inputs 1 and 4? :thinking:

fkzle, 2023-01-04:
Exactly. Input 1-3 can be used independently, since they are all connected to the external diseqc-switch. 

                                                                    Digibit R1
                                  Diseqc-Switch         _________
[LNB 19.2° East ] -------- #### ------------|  In 1   |
                                         ####                 |             |
                                         #### -------------|  In 2    |
                                         ####                 |             |
[LNB 28.5° East ] -------- #### -------------|  In 3    |
                                                                    |              |
[LNB 13° East] ----------------------------------| In 4     |
                                                                     |_______|

Jalle19, 2023-01-05:
Can you try `-A 0:0:0,1:0:1,0:1:0,1:1:1,0:2:0,1:2:1,2:3:0 -S 3:3`?

Jalle19, 2023-01-05:
You could also try `-A 0:0:0,1:0:1,0:1:0,1:1:1,0:2:0,1:2:1,2:3:0 -S 0:0,1:1,2:2,3:3`

fkzle, 2023-01-05:
I already tested the latter 2 days ago. See [Post](https://github.com/Jalle19/satip-axe/issues/53#issuecomment-1370100724)

I will test the other tomorrow, although i doubt that this will help...

Jalle19, 2023-01-05:
I was unsure if you tested with both -A and -S.

fkzle, 2023-01-06:
This did not work as well. 

[image]

Minisatip cannot be convinced to use input 4 if position 2 is requested. 
In addition: if your suggestions with parameter -S would work, this still would be a regression compared to the old minisatip. Before it was possible to watch >1 program on different tuners, if they all are in the same Band. By fixing lnb4 to tuner4, this is not possible anymore.

fkzle, 2023-01-06:
Do you have extended knowledge about the internals of the digibit R1? 
Is the internal switch also addressed via diseqc? or is this done from the axe utils via some internal bus, e.g. i2c?

If it is diseqc, maybe something with -d or -0 could do the magic? If the internal switch is addressed via diseqc, is it committed or uncommitted?

Jalle19, 2023-01-07:
I honestly don't know, sorry.

@catalinii do you know what parameters should be used here?

catalinii, 2023-01-07:
I think 
-A 0:0:0,1:0:1,0:1:0,1:1:1,0:2:0,1:2:1,2:3:0 -S 0:0,1:1,2:2,3:3 should work.

But generally you need a hotbird URL that has src=3 in it and if it does not work upload the logs with 
'-l axe,http -f'

fkzle, 2023-01-07:
@catalinii This is exactly what i tried.
Somehow, the URL with src=3 is internally rewritten to src=1. But Tuner 4 is used. 

If i look at wireshark, src=3 is requested by VDR, but if i then look into the minisatip-webfrontend, i see that tuner 4 is used with the correct frequency and the PIDs requested, but if i look at the link in the web-if, there is src=1 set. 

I will try to upload logs later.

Jalle19, 2023-03-24:
@fkzle did you get it to work?

fkzle, 2023-03-24:
nope. 
after trying a lot of settings and command line parameters, and also after trying to understand the sourcecode, i gave up and am using the old version from perexg.

Jalle19, 2023-03-24:
Could very well be a regression. Closing this though since it doesn't seem to be an issue with the firmware per se.

closed 2023-03-24 by Jalle19