Jalle19/satip-axe #54: Update build environment

Jalle19, 2023-02-14:
The STLinux 2.4 devkit is no longer available (archive.stlinux.com), so the base Docker image this project uses (https://github.com/Jalle19/docker-centos7-stlinux24) can no longer be built. This is a liability.

estimadarocha, 2023-02-14:
Any solution?

Jalle19, 2023-02-14:
I'm working on it

Jalle19, 2023-02-14:
Building with GCC 10 (the default cross-compiler for SH4 available in Debian Bullseye) fails, so that will need updating as well:

```
Failed: -Wl,--start-group  -lm  -Wl,--end-group
Output of:
/usr/bin/sh4-linux-gnu-gcc -Wall -Wshadow -Wwrite-strings -Wundef -Wstrict-prototypes -Wunused -Wunused-parameter -Wunused-function -Wunused-value -Wmissing-prototypes -Wmissing-declarations -Wno-format-security -Wdeclaration-after-statement -Wold-style-definition -fno-builtin-strlen -finline-limit=0 -fomit-frame-pointer -ffunction-sections -fdata-sections -fno-guess-branch-probability -funsigned-char -static-libgcc -falign-functions=1 -falign-jumps=1 -falign-labels=1 -falign-loops=1 -fno-unwind-tables -fno-asynchronous-unwind-tables -fno-builtin-printf -Os -o busybox_unstripped -Wl,--sort-common -Wl,--sort-section,alignment -Wl,--gc-sections -Wl,--start-group applets/built-in.o archival/lib.a archival/libarchive/lib.a console-tools/lib.a coreutils/lib.a coreutils/libcoreutils/lib.a debianutils/lib.a e2fsprogs/lib.a editors/lib.a findutils/lib.a init/lib.a libbb/lib.a libpwdgrp/lib.a loginutils/lib.a mailutils/lib.a miscutils/lib.a modutils/lib.a networking/lib.a networking/libiproute/lib.a networking/udhcp/lib.a printutils/lib.a procps/lib.a runit/lib.a selinux/lib.a shell/lib.a sysklogd/lib.a util-linux/lib.a util-linux/volume_id/lib.a archival/built-in.o archival/libarchive/built-in.o console-tools/built-in.o coreutils/built-in.o coreutils/libcoreutils/built-in.o debianutils/built-in.o e2fsprogs/built-in.o editors/built-in.o findutils/built-in.o init/built-in.o libbb/built-in.o libpwdgrp/built-in.o loginutils/built-in.o mailutils/built-in.o miscutils/built-in.o modutils/built-in.o networking/built-in.o networking/libiproute/built-in.o networking/udhcp/built-in.o printutils/built-in.o procps/built-in.o runit/built-in.o selinux/built-in.o shell/built-in.o sysklogd/built-in.o util-linux/built-in.o util-linux/volume_id/built-in.o -Wl,--end-group -Wl,--start-group -lm -Wl,--end-group
==========
/usr/lib/gcc-cross/sh4-linux-gnu/10/../../../../sh4-linux-gnu/bin/ld: debianutils/lib.a(mktemp.o): in function `mktemp_main':
mktemp.c:(.text.mktemp_main+0xdc): warning: the use of `mktemp' is dangerous, better use `mkstemp' or `mkdtemp'
/usr/lib/gcc-cross/sh4-linux-gnu/10/../../../../sh4-linux-gnu/bin/ld: util-linux/lib.a(rdate.o): in function `rdate_main':
rdate.c:(.text.rdate_main+0x120): undefined reference to `stime'
/usr/lib/gcc-cross/sh4-linux-gnu/10/../../../../sh4-linux-gnu/bin/ld: coreutils/lib.a(date.o): in function `date_main':
date.c:(.text.date_main+0x2d4): undefined reference to `stime'
collect2: error: ld returned 1 exit status
Note: if build needs additional libraries, put them in CONFIG_EXTRA_LDLIBS.
Example: CONFIG_EXTRA_LDLIBS="pthread dl tirpc audit pam"
make[1]: *** [Makefile:717: busybox_unstripped] Error 1
make[1]: Leaving directory '/build/apps/busybox-1.26.2'
make: *** [Makefile:348: apps/busybox-1.26.2/busybox] Error 2
make: *** [Makefile:97: docker-release] Error 2
```

Jalle19, 2023-02-17:
Made some progress today after being stuck on a Debian bug for a few days. Seems like cross-compilation only works in Stretch, not newer versions. That's still years fresher than the STLinux 2.4 disitribution though.

Some packages need updates in order to compile, but so far no major blockers.

estimadarocha, 2023-02-17:
So will we have a new kernel???

Jalle19, 2023-02-18:
No

Jalle19, 2023-02-19:
Got everything to build, but can't assemble the final root filesystem since we don't have the various SH4 compiled .rpm packages available anymore...

It seems like the only sane option is to completely rewrite the build system with Buildroot, however that's no easy task either.

estimadarocha, 2023-04-17:
@Jalle19 you discovered why the last versiosn were not working and you had to revert to v1.2.12?

Jalle19, 2023-04-17:
Not really, I wanted to get a new official build out without worrying about minisatip. Let me know if there's a newer version that definitively works!

estimadarocha, 2023-04-17:
the last time i tested at least for me when trying to pull pids = all it was a subscription error it was 8191 and not 8192! but i think @catalinii solve it.

Jalle19, 2023-05-31:
Will need to make a custom Docker image that has the SH4 toolchain and at least libssl (needed for oscam), zlib (needed for Dropbear) and ncurses (needed for nano). After that it should be possible to drastically simplify do_min_fs.py and be able to move off the current build environment conpletely.

I dabbled with buildroot and got the kernel and base image built, but it was quite the hassle and too unfamiliar for now.

Jalle19, 2023-06-29:
Getting the kernel to compile with GCC 8 turned out to be quite the chore. The bundled version in this repository just wouldn't compile no matter what, but the latest upstream 2.6.32.71 version compiled nicely after some obligatory patches.

Jalle19, 2024-07-04:
Some progress made in https://github.com/Jalle19/satip-axe/tree/external-kernel, will need to set up serial access to one of my devices so I can debug the image and get it to run.

estimadarocha, 2024-08-10:
@Jalle19 any plans for new release??

Jalle19, 2024-08-10:
Not at the moment

ast261, 2025-06-17:
Will the axe platform be removed from minisatip v2 ?
see: https://github.com/catalinii/minisatip/commit/a0cc71d00244111104e7ce21e4de1616b64ef6ca

ast261, 2025-06-19:
There is no need for removal.

I have made two toolchains:
gcc 4.9.4 to compile the kernel and a modern C++ compiler (gcc 10) for all userspace components (https://github.com/ast261/sh4-buildroot-glibc-2.23/releases). 
Both are based on glibc 2.23, the last compatible version for our kernel.

First tests with upcoming v2 look promising (https://github.com/ast261/minisatip/commits/idl4k-v2). 
Axe is alive!

catalinii, 2025-06-19:
There are few reasons:
- old hardware with low cpu resources
-  not easily available anymore
- minisatip 1.x still supports it, sowne can backport fixes of they help AXE use case
- maintaining a platform is costly

I think there are a lot of compromises that we need to add to minisatip 2.x to support AXE going forward with no benefits due to low CPU power, old kernel, old glibc, old libc++.

If you look back since adding AXE we managed to mess the switch configuration with no other real improvements.

ast261, 2025-06-19:
As an outsider, I am not aware of compromises.
Thanks for explanation.