Jalle19/satip-axe #56: minisatip (24 rc3) restart automatically at the same hour all days?

roisiano, 2023-02-17:
Hello:

I'm using 24 rc3 version.

I have noticed since a couple of weeks, more or less, that, at a certain time (in this case, more or less at 20:04:xy), I lose the connection with the TV channels that I'm watching at that time.
After 10-20 seconds, it recovers.
All days at that time. Always at that hour and only at that time.

Today, I programmed a couple of recordings with tvheadend for starting at 20:03:30 and ending at 20:06:30, and I was watching a live channel too at the same time.
I was watching tvheadend interface, the TV-Box interface (CoreELEC server with tvheadend), and doing pings too, to try to "investigate" what could be the problem.

OK; this is tvheadend log (I edited users, channel names, etc.):
`
2023-02-17 20:04:50.451 satip: SAT>IP DVB-S Tuner #1 (192.168.1.60@TCP) - RTSP error -5 (Input/output error) [9874-0]

2023-02-17 20:04:50.464 satip: SAT>IP DVB-S Tuner #2 (192.168.1.60@TCP) - RTSP error -104 (Connection reset by peer) [5-0]

2023-02-17 20:04:50.465 satip: SAT>IP DVB-S Tuner #3 (192.168.1.60@TCP) - RTSP error -5 (Input/output error) [9874-0]

2023-02-17 20:04:52.013 subscription: 0019: service instance is bad, reason: Tuning failed

2023-02-17 20:04:52.014 mpegts: 11973V in Astra 23.5E - tuning on SAT>IP DVB-S Tuner #4 (192.168.1.60@TCP)

2023-02-17 20:04:52.014 capmt: oscamuser: Starting CAPMT server for service "Channel 2 HD" on adapter 1

2023-02-17 20:04:52.014 subscription: 0019: "DVR: " subscribing on channel "Channel 2 HD", weight: 500, adapter: "SAT>IP DVB-S Tuner #4 (192.168.1.60@TCP)", network: "Astra 23.5E", mux: "11973V", provider: "Group", service: "Channel 2 HD", profile="pass", username="user"

2023-02-17 20:04:52.014 subscription: 0018: service instance is bad, reason: Tuning failed

2023-02-17 20:04:52.015 mpegts: 10817.5V in Astra Manual - tuning on SAT>IP DVB-S Tuner #2 (192.168.1.60@TCP)

2023-02-17 20:04:52.015 capmt: oscamuser: Starting CAPMT server for service "Channel 1" on adapter 2

2023-02-17 20:04:52.015 subscription: 0018: "DVR: " subscribing on channel "Channel 1", weight: 500, adapter: "SAT>IP DVB-S Tuner #2 (192.168.1.60@TCP)", network: "Astra Manual", mux: "10817.5V", provider: "+", service: "Channel 1", profile="pass", username="user"

2023-02-17 20:04:52.015 subscription: 0017: service instance is bad, reason: Tuning failed

2023-02-17 20:04:52.015 satip: SAT>IP DVB-S Tuner #4 (192.168.1.60@TCP) - RTSP error -111 (Connection refused) [6-0]

2023-02-17 20:04:52.016 mpegts: 10935.5V in Astra Manual - tuning on SAT>IP DVB-S Tuner #3 (192.168.1.60@TCP)

2023-02-17 20:04:52.016 capmt: oscamuser: Starting CAPMT server for service "Channel 3" on adapter 0

2023-02-17 20:04:52.016 satip: SAT>IP DVB-S Tuner #2 (192.168.1.60@TCP) - RTSP error -111 (Connection refused) [6-0]

2023-02-17 20:04:52.016 subscription: 0017: "127.0.0.1 [ user2 | Kodi Media Center ]" subscribing on channel "Channel 3", weight: 100, adapter: "SAT>IP DVB-S Tuner #3 (192.168.1.60@TCP)", network: "Astra Manual", mux: "10935.5V", provider: "+", service: "Channel 3", profile="htsp", hostname="127.0.0.1", username="user2", client="Kodi Media Center"

2023-02-17 20:04:52.017 satip: SAT>IP DVB-S Tuner #3 (192.168.1.60@TCP) - RTSP error -111 (Connection refused) [6-0]

2023-02-17 20:04:52.025 satip: SAT>IP DVB-S Tuner #4 (192.168.1.60@TCP) - failed to tune (-5)

2023-02-17 20:04:52.026 satip: SAT>IP DVB-S Tuner #2 (192.168.1.60@TCP) - failed to tune (-5)

2023-02-17 20:04:52.064 satip: SAT>IP DVB-S Tuner #4 (192.168.1.60@TCP) - failed to tune (-5)

2023-02-17 20:04:52.064 satip: SAT>IP DVB-S Tuner #2 (192.168.1.60@TCP) - failed to tune (-5)

2023-02-17 20:04:52.064 satip: SAT>IP DVB-S Tuner #3 (192.168.1.60@TCP) - failed to tune (-5)

2023-02-17 20:04:54.013 subscription: 0019: service instance is bad, reason: Tuning failed

2023-02-17 20:04:54.013 subscription: 0018: service instance is bad, reason: Tuning failed

2023-02-17 20:04:54.014 mpegts: 10817.5V in Astra Manual - tuning on SAT>IP DVB-S Tuner #1 (192.168.1.60@TCP)

2023-02-17 20:04:54.014 capmt: oscamuser: Starting CAPMT server for service "Channel 1" on adapter 1

2023-02-17 20:04:54.014 subscription: 0018: "DVR: " subscribing on channel "Channel 1", weight: 500, adapter: "SAT>IP DVB-S Tuner #1 (192.168.1.60@TCP)", network: "Astra Manual", mux: "10817.5V", provider: "+", service: "Channel 1", profile="pass", username="user"

2023-02-17 20:04:54.014 subscription: 0017: service instance is bad, reason: Tuning failed
[... 169 lines cut ...]
#
# inetd (httpd,ftpd)
#
INETD="yes"

# crond
#
CROND="yes"

#
# nfsd v3
#
NFSD="yes"
NFSD_HOTPLUG_EXPORT="*" # replace with host or subnet, '*' = world

#
# oscamd
#
OSCAMD="no"

#
# custom daemon/service/setup executed from sysinit
#
#CUSTOMD="/etc/sysconfig/myboot"
`

Jalle19, 2023-02-17:
Can you post the output of `logread` from your SAT>IP server?

roisiano, 2023-02-17:
> Can you post the output of `logread` from your SAT>IP server?

It seems the internal date of the receiver is NOT "real", but, a DHCP deconfig all days at 00:00 (perhaps "real time" 20:04:xy)?

Perhaps a problem with the router?

> Jan 10 00:00:07 (none) local0.notice root: DHCP bound ('192.168.1.60' 'broadcast 192.168.1.255' 'netmask 255.255.255.0' | '192.168.1.1' '80.58.61.250 80.58.61.254') replacing ''
> Jan 10 00:00:17 (none) user.notice minisatip: Setting jess adapter 0 slot 0 freq 1210 pin -1
> Jan 10 00:00:17 (none) user.notice minisatip: Setting jess adapter 2 slot 1 freq 1420 pin -1
> Jan 10 00:00:17 (none) user.notice minisatip: Set timeout for adapter 0 to 600000
> Jan 10 00:00:17 (none) user.notice minisatip: Set timeout for adapter 2 to 600000
> Jan 10 00:00:17 (none) user.notice minisatip: Setting diseqc adapter 1 fast 0 addr 0x10 committed_no 4 uncommitted_no 0
> Jan 10 00:00:17 (none) user.notice minisatip: Setting diseqc adapter 3 fast 0 addr 0x10 committed_no 4 uncommitted_no 0
> Jan 10 00:00:17 (none) daemon.notice minisatip[1558]: minisatip version 1.2.12, compiled in Dec 19 2022 21:14:25, with s2api version: 0501
> Jan 10 00:00:17 (none) daemon.notice minisatip[1558]: Initializing with 4 devices
> Jan 10 04:52:17 (none) daemon.notice minisatip[1558]: Maximum pid filter reached, lowering the value to 32
> Jan 10 06:00:07 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 09:00:07 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 10:30:07 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:15:07 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:37:37 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:48:52 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:54:30 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:57:18 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:58:42 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:59:24 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:59:45 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 11:59:54 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 12:00:00 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 12:00:02 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 12:00:03 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 12:00:03 (none) daemon.err udhcpc[726]: lease lost, entering init state
> Jan 10 12:00:03 (none) local0.notice root: DHCP deconfig
> Jan 10 12:00:03 (none) daemon.err udhcpc[726]: sending discover
> Jan 10 12:00:03 (none) local0.notice root: minisatip exited 137, restarting
> Jan 10 12:00:04 (none) daemon.err udhcpc[726]: sending select for 192.168.1.60
> Jan 10 12:00:05 (none) daemon.err udhcpc[726]: lease of 192.168.1.60 obtained, lease time 43200
> Jan 10 12:00:05 (none) local0.notice root: DHCP bound ('192.168.1.60' 'broadcast 192.168.1.255' 'netmask 255.255.255.0' | '192.168.1.1' '80.58.61.250 80.58.61.254') replacing ''
> Jan 10 12:00:15 (none) user.notice minisatip: Setting jess adapter 0 slot 0 freq 1210 pin -1
> Jan 10 12:00:15 (none) user.notice minisatip: Setting jess adapter 2 slot 1 freq 1420 pin -1
> Jan 10 12:00:15 (none) user.notice minisatip: Set timeout for adapter 0 to 600000
> Jan 10 12:00:15 (none) user.notice minisatip: Set timeout for adapter 2 to 600000
> Jan 10 12:00:15 (none) user.notice minisatip: Setting diseqc adapter 1 fast 0 addr 0x10 committed_no 4 uncommitted_no 0
> Jan 10 12:00:15 (none) user.notice minisatip: Setting diseqc adapter 3 fast 0 addr 0x10 committed_no 4 uncommitted_no 0
> Jan 10 12:00:15 (none) daemon.notice minisatip[1589]: minisatip version 1.2.12, compiled in Dec 19 2022 21:14:25, with s2api version: 0501
> Jan 10 12:00:16 (none) daemon.notice minisatip[1589]: Initializing with 4 devices
> Jan 10 18:00:06 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 19:04:09 (none) daemon.notice minisatip[1589]: Maximum pid filter reached, lowering the value to 32
> Jan 10 21:00:06 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 22:30:07 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:15:07 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:37:37 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:48:52 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:54:29 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:57:17 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:58:41 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:59:24 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:59:45 (none) daemon.err udhcpc[726]: sending renew
> Jan 10 23:59:55 (none) daemon.err udhcpc[726]: sending renew
> Jan 11 00:00:00 (none) daemon.err udhcpc[726]: sending renew
> Jan 11 00:00:02 (none) daemon.err udhcpc[726]: sending renew
> Jan 11 00:00:02 (none) daemon.err udhcpc[726]: sending renew
> Jan 11 00:00:02 (none) daemon.err udhcpc[726]: lease lost, entering init state
> Jan 11 00:00:02 (none) local0.notice root: DHCP deconfig
> Jan 11 00:00:02 (none) daemon.err udhcpc[726]: sending discover
> Jan 11 00:00:02 (none) local0.notice root: minisatip exited 137, restarting
> Jan 11 00:00:04 (none) daemon.err udhcpc[726]: sending select for 192.168.1.60
> Jan 11 00:00:05 (none) daemon.err udhcpc[726]: lease of 192.168.1.60 obtained, lease time 43200
> Jan 11 00:00:05 (none) local0.notice root: DHCP bound ('192.168.1.60' 'broadcast 192.168.1.255' 'netmask 255.255.255.0' | '192.168.1.1' '80.58.61.250 80.58.61.254') replacing ''
[... cut ...]
> Jan 14 12:19:08 (none) authpriv.info dropbear[1847]: Child connection from 192.168.1.40:61636
> Jan 14 12:19:11 (none) authpriv.notice dropbear[1847]: Password auth succeeded for 'root' from 192.168.1.40:61636
> Jan 14 13:06:16 (none) authpriv.info dropbear[1847]: Exit (root) from <192.168.1.40:61636>: Disconnect received
> Jan 14 13:08:16 (none) authpriv.info dropbear[1865]: Child connection from 192.168.1.40:16723
> Jan 14 13:08:22 (none) authpriv.notice dropbear[1865]: Password auth succeeded for 'root' from 192.168.1.40:16723
> ~ #

roisiano, 2023-02-17:
Yes. It seems that the problem was in the router.
I had configured a static lease (192.168.1.60) for Digibit R1 but it seems that (perhaps a router firmware update?) it was limited to 12 hours (43200 seconds).

Now I have changed it in router config, and it seems that it should work:

> 
> ~ # logread
> Jan  1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.26.2
> Jan  1 12:00:00 (none) local0.notice root: AXE drivers load start
> Jan  1 12:00:01 (none) local0.notice root: AXE drivers load finish
> Jan  1 12:00:01 (none) daemon.info init: starting pid 726, tty '': '/etc/init.d/network'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 727, tty '': '/etc/init.d/ntpd'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 728, tty '': '/etc/init.d/nfsd'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 729, tty '': '/etc/init.d/satip'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 730, tty '': '/etc/init.d/telnetd'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 731, tty '': '/etc/init.d/dropbear'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 732, tty '': '/etc/init.d/inetd'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 734, tty '': '/etc/init.d/crond'
> Jan  1 12:00:01 (none) daemon.info init: starting pid 736, tty '': '/etc/init.d/oscamd'
> Jan  1 12:00:01 (none) daemon.err udhcpc[726]: started, v1.26.2
> Jan  1 12:00:01 (none) daemon.info init: starting pid 738, tty '': '/etc/init.d/customd'
> Jan  1 12:00:01 (none) local0.notice root: DHCP deconfig
> Jan  1 12:00:02 (none) authpriv.warn dropbear[731]: Failed loading /etc/dropbear/dropbear_dss_host_key
> Jan  1 12:00:02 (none) authpriv.warn dropbear[731]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
> Jan  1 12:00:02 (none) authpriv.warn dropbear[731]: Failed loading /etc/dropbear/dropbear_ed25519_host_key
> Jan  1 12:00:02 (none) authpriv.info dropbear[731]: Not backgrounding
> Jan  1 12:00:02 (none) cron.info crond[734]: crond (busybox 1.26.2) started, log level 8
> Jan  1 12:00:02 (none) local0.notice root: Setting STV6210 gain to 8
> Jan  1 12:00:02 (none) daemon.warn mountd[728]: Kernel does not have pseudo root support.
> Jan  1 12:00:02 (none) daemon.warn mountd[728]: NFS v4 mounts will be disabled unless fsid=0
> Jan  1 12:00:02 (none) daemon.warn mountd[728]: is specfied in /etc/exports file.
> Jan  1 12:00:02 (none) daemon.notice mountd[728]: Version 1.3.3 starting
> Jan  1 12:00:02 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:00:05 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:00:05 (none) local0.notice root: Mounting sda : /media/sda
> Jan  1 12:00:05 (none) local0.notice root: Mounting sda1 : /media/sda1
> Jan  1 12:00:05 (none) local0.notice root: ... sda1 mounted using vfat filesystem (nfsd exported)
> Jan  1 12:00:08 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:00:11 (none) local0.notice root: DHCP leasefail
> Jan  1 12:00:30 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:00:33 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:00:36 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:00:39 (none) local0.notice root: DHCP leasefail
> Jan  1 12:01:00 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:01:03 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:01:06 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:01:09 (none) local0.notice root: DHCP leasefail
> Jan  1 12:01:28 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:01:32 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:01:35 (none) daemon.err udhcpc[726]: sending discover
> Jan  1 12:01:37 (none) local0.notice root: DHCP leasefail
> Jan  1 12:01:57 (none) daemon.err udhcpc[726]: sending discover
**> Jan  1 12:02:00 (none) daemon.err udhcpc[726]: sending select for 192.168.1.60
> Jan  1 12:02:01 (none) daemon.err udhcpc[726]: lease of 192.168.1.60 obtained, lease time 268435455**
> Jan  1 12:02:01 (none) local0.notice root: DHCP bound ('192.168.1.60' 'broadcast 192.168.1.255' 'netmask 255.255.255.0' | '192.168.1.1' '80.58.61.250 80.58.61.254') replacing ''
> Jan  1 12:02:11 (none) user.notice minisatip: Setting jess adapter 0 slot 0 freq 1210 pin -1
> Jan  1 12:02:11 (none) user.notice minisatip: Setting jess adapter 2 slot 1 freq 1420 pin -1
> Jan  1 12:02:11 (none) user.notice minisatip: Set timeout for adapter 0 to 600000
> Jan  1 12:02:11 (none) user.notice minisatip: Set timeout for adapter 2 to 600000
> Jan  1 12:02:11 (none) user.notice minisatip: Setting diseqc adapter 1 fast 0 addr 0x10 committed_no 4 uncommitted_no 0
> Jan  1 12:02:11 (none) user.notice minisatip: Setting diseqc adapter 3 fast 0 addr 0x10 committed_no 4 uncommitted_no 0
> Jan  1 12:02:11 (none) daemon.notice minisatip[1037]: minisatip version 1.2.12, compiled in Dec 19 2022 21:14:25, with s2api version: 0501
> Jan  1 12:02:12 (none) daemon.notice minisatip[1037]: Initializing with 4 devices
> Jan  1 12:04:21 (none) authpriv.info dropbear[1038]: Child connection from 192.168.1.40:1085
> Jan  1 12:04:26 (none) authpriv.notice dropbear[1038]: Password auth succeeded for 'root' from 192.168.1.40:1085
> ~ # 

268435455 seconds (> 3000 years) of leasing should be enough for me. 

I think you can delete the issue. Sorry for the inconvenience and kind regards.

Jalle19, 2023-02-17:
No problem, glad you found the issue!

closed 2023-02-17 by Jalle19