Jalle19/satip-axe #61: Minisatip glitches out after around a month of uptime and eventually whole box hangs (no response on networking)

rajkosto, 2023-04-16:
After 27d it starts glitching out like this: 
https://images.sshnuke.net/2023-04-16_05-03-07_l7T8OSB9A.png
https://images.sshnuke.net/2023-04-16_05-03-24_hrBaQafst.png
(minisatip wont load the json data needed for its table to show up, streams still working but won't show Signal Level/SNR in tvheadend)

```
Welcome to SATIP-AXE server 202212181117-24-rc2
~ # uptime
 05:04:09 up 27 days, 11:12, 1 users, load average: 9.99, 9.97, 9.90
```

After a few more days of running like this it will completely stop responding on network and the only solution is to physically turn it off and then on again.
There is nothing strange in top CPU usage/process list or in dmesg when it's "glitched" like this vs when just operating normally
If it's glitched, just doing a reboot in ssh fixes it, to be preemptive and not have to access the box physically to restart it when it breaks completely.

Jalle19, 2023-04-16:
Could be a memory leak, but I haven't noticed this problem personally. Can you post the output of `logread`?

rajkosto, 2023-05-11:
It has been 24d of uptime and it's glitching out again
https://images.sshnuke.net/2023-05-11_12-17-06_5zZIyaQ5s.png (strangely it usually shows unknown on all the TPs im using, not just one)
of course, minisatip webpage will not open past this now: https://images.sshnuke.net/2023-05-11_12-19-07_LSNctHbiU.png

what exactly did you want from logs ?

Jalle19, 2023-05-11:
It seems like your router has been hacked and is actively trying to force its way in to your SAT>IP device, hence all the invalid attempts. I suggest you fix that ASAP.

Jalle19, 2023-05-11:
Could be that Dropbear (the SSH daemon) uses up all process IDs or ports or something like that 🤷 see if the problem goes away once you fix your router (192.168.1.1).

Jalle19, 2023-05-11:
You can find the process ID using `ps -ef | grep minisatip`, then `kill 123` assuming 123 is the PID. It should restart itself after a few seconds.

rajkosto, 2023-05-11:
It did restart, and now its showing its webpage properly: https://images.sshnuke.net/2023-05-12_01-04-36_Bo1uuVnsR.png
(interestingly none of the counters have restarted from 0)
TVHeadend can now also query the signal information as well: https://images.sshnuke.net/2023-05-12_01-05-21_vc3cEg4wm.png

Jalle19, 2023-05-26:
I'm closing this for now, feel free to reopen it if you get the box to lock up completely

closed 2023-05-26 by Jalle19

rajkosto, 2023-06-30:
After running for another 47 days with only the minisatip restarted with kill, minisatip is again in it's "bugged" state where it wont show any signal information either via satip or via its webpage.
Obviously this time nothing has been attacking the box via ssh because those port forwards are still disabled.
So what to do about this now ?

Jalle19, 2023-07-01:
So are you saying minisatip works fine except there is no signal information anymore? I'm afraid that's an old bug and I don't have a solution for it.

Does the web interface work at all?

rajkosto, 2023-07-01:
the minisatip httpd works to serve its static content, but not the json file (just no reply from server on that one, waiting for), therefore the website looks like this: https://images.sshnuke.net/2023-04-16_05-03-07_l7T8OSB9A.png

Jalle19, 2023-07-01:
@catalinii any idea what to do to debug this?

catalinii, 2023-07-01:
https://medium.com/@sourabhedake/core-dumps-how-to-enable-them-73856a437711

This should allow killing minisatip with -ABRT:
kill -s ABRT <minisatip_pid>

rajkosto, 2023-07-12:
minisatip is now showing 0% signal/snr in both tvheadend and on its own page:
https://images.sshnuke.net/2023-07-12_22-44-09_jekJtggnK.png