Jalle19/satip-axe #62: energy saving

Lip90, 2023-04-21:
What are the best energy saving settings? I have two satellite cables connected which come from a multiswitch. The question may seem silly but somehow I do not understand the documentary.

Jalle19, 2023-04-21:
You don't need to do anything special, the tuners will be powered down after some time of inactivity.

Lip90, 2023-04-23:
I thought you have to deposit such a parameter? is not necessary?
`#
# minisatip 1.0+
#
MINISATIP="yes"
MINISATIP_OPTS="-W 0"
`

Jalle19, 2023-04-23:
I'm not entirely sure, I haven't used -W personally at least

Lip90, 2023-04-23:
I got the documentation from [here](https://github.com/perexg/satip-axe/blob/master/dist/minisatip.md). But I'm not sure if I understood it correctly. I thought without these paramenter a standby is not possible?

Jalle19, 2023-04-25:
The tuners will power down regardless of which -W mode you use.

closed 2023-04-25 by Jalle19