Jalle19/satip-axe #64: diseqc problems.

misushot, 2023-04-22:
hi i use -A 0:0:0,1:0:1,2:0:2,3:0:3,4:1:0,5:1:1,6:1:2,7:1:3 for my virtual-diseqc setup for 2x4positions but i see only lnb1 for both tuners and only tuner1 works. i have seen a --dmx-source 0:0,1:1 option for coax1(lnb1) for tuner1 and coax2(lnb2) for tuner2. with fe variable and without -A in url it works perfectly.

Jalle19, 2023-04-25:
Can you explain what you're trying to accomplish? I'm having trouble understanding, especially your last sentence.

misushot, 2023-04-25:
ok i use two tuners for 8 sat positions.
tuner1 -> Diseqc 1,2,3,4
tuner2 -> Diseqc 1,2,3,4

-A 0:0:0,1:0:1,2:0:2,3:0:3,4:1:0,5:1:1,6:1:2,7:1:3 is the commandline to seperate this src positions.

i will use this tuners without "fe" like:
http://x.x.x.x/?src=1&freq=xxxxx...
http://x.x.x.x/?src=2&freq=xxxxx...
http://x.x.x.x/?src=3&freq=xxxxx...
http://x.x.x.x/?src=4&freq=xxxxx...
http://x.x.x.x/?src=5&freq=xxxxx...
http://x.x.x.x/?src=6&freq=xxxxx...
http://x.x.x.x/?src=7&freq=xxxxx...
http://x.x.x.x/?src=8&freq=xxxxx...

src 1-4 is on tuner1
src 5-8 is on tuner2

src 1-4 works but 4-8 have problems and connect to (lnb1,coax) and not to (lnb2,coax2)

misushot, 2023-04-25:
with fe variable it works.
http://x.x.x.x/?fe=1&src=1&freq=xxxxx...
http://x.x.x.x/?fe=1&src=2&freq=xxxxx...
http://x.x.x.x/?fe=1&src=3&freq=xxxxx...
http://x.x.x.x/?fe=1&src=4&freq=xxxxx...
http://x.x.x.x/?fe=2&src=1&freq=xxxxx...
http://x.x.x.x/?fe=2&src=2&freq=xxxxx...
http://x.x.x.x/?fe=2&src=3&freq=xxxxx...
http://x.x.x.x/?fe=2&src=4&freq=xxxxx...

but i need this without fe variable

Jalle19, 2023-05-26:
Please try `-A 0:0:0,1:0:1,2:0:2,3:0:3,4:1:0,5:1:1,6:1:2,7:1:3 -S 0-3:0,4-7:1`

Jalle19, 2023-05-29:
Closing this, please open an issue at https://github.com/catalinii/minisatip if the above trick doesn't work

closed 2023-05-29 by Jalle19