Jalle19/satip-axe #65: /usr/sbin/ntpd is started during boot up, keeps running but doesn't adjust clock

phaze75, 2023-04-29:
Hi @Jalle19,
I updated recently to your build 24, thanks again for the hint concerning the USB upgrade procedure. There is another issue, that exists for quite some time. The clock of the Digibit R1 box (only this box?) does not get updated by ntpd. The daemon is started properly during boot up and running in the background:

~ # ps x | grep ntpd
  729 root       0:00 {ntpd} /bin/sh /etc/init.d/ntpd
  986 root       0:06 /usr/sbin/ntpd -n -q -I eth0 -p 0.de.pool.ntp.org

However, it does not update the time of the box. Not until it is started with the exact same parameters manually from the command line:

~ # /usr/sbin/ntpd -n -q -I eth0 -p 0.de.pool.ntp.org
ntpd: setting time to 2023-04-29 17:39:53.857995 (offset +262635664.247929s)

Do you have any suggestions on how to fix this?

Best regards!

Jalle19, 2023-05-01:
Was there nothing in the log after startup? It should at least sync the time once.

phaze75, 2023-05-04:
No, I am afraid not. That is the log dump after a reboot:

~ # logread
Jan  1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.26.2
Jan  1 12:00:00 (none) local0.notice root: AXE drivers load start
Jan  1 12:00:01 (none) local0.notice root: AXE drivers load finish
Jan  1 13:00:01 (none) local0.notice root: ttyUSB0 device attached '4-1:1.0'
Jan  1 12:00:01 (none) daemon.info init: starting pid 728, tty '': '/etc/init.d/network'
Jan  1 12:00:01 (none) daemon.info init: starting pid 729, tty '': '/etc/init.d/ntpd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 731, tty '': '/etc/init.d/nfsd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 732, tty '': '/etc/init.d/satip'
Jan  1 12:00:01 (none) daemon.info init: starting pid 733, tty '': '/etc/init.d/telnetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 734, tty '': '/etc/init.d/dropbear'
Jan  1 12:00:01 (none) daemon.info init: starting pid 735, tty '': '/etc/init.d/inetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 736, tty '': '/etc/init.d/crond'
Jan  1 13:00:01 (none) daemon.err udhcpc[728]: started, v1.26.2
Jan  1 12:00:02 (none) daemon.info init: starting pid 737, tty '': '/etc/init.d/oscamd'
Jan  1 12:00:02 (none) daemon.info init: starting pid 742, tty '': '/etc/init.d/customd'
Jan  1 13:00:02 (none) local0.notice root: DHCP deconfig
Jan  1 13:00:02 (none) local0.notice root: ttyUSB0 device attached '4-1:1.0'
Jan  1 13:00:02 (none) authpriv.warn dropbear[734]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan  1 13:00:02 (none) authpriv.warn dropbear[734]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan  1 13:00:02 (none) authpriv.warn dropbear[734]: Failed loading /etc/dropbear/dropbear_ed25519_host_key
Jan  1 13:00:02 (none) authpriv.info dropbear[734]: Not backgrounding
Jan  1 13:00:02 (none) cron.info crond[736]: crond (busybox 1.26.2) started, log level 8
Jan  1 13:00:02 (none) local0.notice root: Setting STV6210 gain to 8
Jan  1 13:00:02 (none) daemon.warn mountd[731]: Kernel does not have pseudo root support.
Jan  1 13:00:02 (none) daemon.warn mountd[731]: NFS v4 mounts will be disabled unless fsid=0
Jan  1 13:00:02 (none) daemon.warn mountd[731]: is specfied in /etc/exports file.
Jan  1 13:00:02 (none) daemon.notice mountd[731]: Version 1.3.3 starting
Jan  1 13:00:02 (none) daemon.err udhcpc[728]: sending discover
Jan  1 13:00:05 (none) daemon.err udhcpc[728]: sending discover
Jan  1 13:00:08 (none) daemon.err udhcpc[728]: sending discover
Jan  1 13:00:11 (none) local0.notice root: DHCP leasefail
Jan  1 13:00:31 (none) daemon.err udhcpc[728]: sending discover
Jan  1 13:00:31 (none) daemon.err udhcpc[728]: sending select for 192.168.0.112
Jan  1 13:00:31 (none) daemon.err udhcpc[728]: lease of 192.168.0.112 obtained, lease time 7200
Jan  1 13:00:31 (none) local0.notice root: DHCP bound ('192.168.0.112' '' 'netmask 255.255.255.0' | '192.168.0.254' '192.168.0.254') replacing ''
Jan  1 13:00:41 (none) daemon.notice minisatip[1007]: minisatip version 1.2.12, compiled in Mar 24 2023 16:49:12, with s2api version: 0501
Jan  1 13:00:41 (none) daemon.notice minisatip[1007]: Initializing with 4 devices

phaze75, 2023-05-05:
> The problem seems to be that your device is not getting a DHCP lease immediately: 

I agree and I have realized this earlier, but I cannot figure out why this happens. I use OPNsense as a firewall and DHCP/DNS server and there aren't any issues with other clients. Further, there is a DHCP static IP mapping configured, that uses satip's MAC address. I don't see how I could improve this.

I just restarted the satip box to get the debug log on the DHCP server side:

2023-05-05T10:29:18	Informational	dhcpd	 DHCPACK on 192.168.0.112 to 00:24:a6:00:25:6e via igb1
2023-05-05T10:29:18	Informational	dhcpd	 DHCPREQUEST for 192.168.0.112 (192.168.0.254) from 00:24:a6:00:25:6e via igb1
2023-05-05T10:29:18	Informational	dhcpd	 DHCPOFFER on 192.168.0.112 to 00:24:a6:00:25:6e via igb1
2023-05-05T10:29:18	Informational	dhcpd	 DHCPDISCOVER from 00:24:a6:00:25:6e via igb1

So, imho there isn't any hint for a DHCP issue existing on the server side. Might there be a possibility that the client side could be fishy in some way?

mark-royds, 2023-05-05:
> So, imho there isn't any hint for a DHCP issue existing on the server side. Might there be a possibility that the client side could be fishy in some way?
Hi.  Are you connecting the SatIP box directly to your OPNsense firewall, or is there an Ethernet switch in between?
I've seen similar delays in obtaining a DHCP lease where a device attached to an Ethernet switch running Spanning Tree Protocol is blocked for several seconds after the link comes up while STP does its thing.

Jalle19, 2023-05-05:
FWIW on my box with 12 days uptime `ntpd` is no longer running, so there's definitely some improvements to be made here still.

phaze75, 2023-05-05:
>Hi. Are you connecting the SatIP box directly to your OPNsense firewall, or is there an Ethernet switch in between?

Hi! There are two Ethernet Gigabit switches (unmanaged) in between. But there is no way I could change that physically. I see your point, but I guess the DHCP service dependencies should be robust enough to handle that.

mark-royds, 2023-05-05:
> There are two Ethernet Gigabit switches (unmanaged) in between
In my experience unmanaged switches don't do STP so likely not the cause of your issue... Ignore me 😁
Other comments here suggest ntpd is quite sensitive so perhaps some improvements to be made.

In any case, my ntpd seems quite stable:
`~ # ps -a | grep ntp
 3148 root       0:00 /usr/sbin/ntpd -I eth0 -p 0.pool.ntp.org
 5293 root       0:00 grep ntp
~ # uptime
 23:52:26 up 40 days, 11:37,  1 users,  load average: 6.01, 6.00, 6.00
~ #`

Jalle19, 2023-05-06:
It used to be quite stable, but we recently did some changes to ensure that time was synced before minisatip started. The problems seem to have started then.

phaze75, 2023-05-06:
It‘s great that you work on it. I had a discussion with @perexg some time ago, but it was never fixed. There has also been an an issue with using FQDNs instead of IPs for ntpd time servers, because of DNS resolving issues.

rajkosto, 2023-05-11:
> Hi! There are two Ethernet Gigabit switches (unmanaged) in between. But there is no way I could change that physically. I see your point, but I guess the DHCP service dependencies should be robust [...]

When i had the satip box connected to a realtek chipset unmanaged switch, it would sometimes have to try getting the Ethernet link up like 7 times before it would stabilise (you could see the Ethernet port turning off after 2 seconds, turning back on, and repeating, the same would show up in dmesg as link up link down) during this time, the 10s window to get ntp has passed. I've since replaced that switch with a brand new 10$ ECONET chipset one, and on that one it links up immediately the first time, every time, at 1000mb/s (on the old one changing cables and cleaning the ports didnt make a difference, the satip just didn't like that switch chipset)

phaze75, 2023-05-11:
>When i had the satip box connected to a Realtek chipset unmanaged switch, it would sometimes have to try getting the

Hi, I just checked. The D-Link switch directly connected to my Digibit R1 box has indeed a Realtek 8370N chipset. The main switch is a HP ProCurve with no info about the chipset.
Anyway, even if the Realtek hypothesis would be true, a quick and dirty workaround could be a configurable time(out) window (currently fixed to 10s) in the startup script.

Jalle19, 2023-05-24:
Can't find a reason for why `ntpd` exits randomly. I guess a decent option would be to compile `chronyd` and switch to that. It doesn't need hacks to wait for DNS being available, and can run in the foreground so we can use a simple loop in the init script to keep it running in case it exits (just like we do for minisatip).

closed 2023-05-29 by Jalle19

Jalle19, 2023-06-27:
I've published a pre-release containing `chronyd` here: https://github.com/Jalle19/satip-axe/releases/tag/build25-rc1

PR #66 (2023-05-29): Use chronyd instead of BusyBox's ntpd
Should fix #65 

```
~ # logread
Jan  1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.26.2
Jan  1 12:00:00 (none) local0.notice root: AXE drivers load start
Jan  1 12:00:01 (none) local0.notice root: AXE drivers load finish
Jan  1 12:00:01 (none) daemon.info init: starting pid 718, tty '': '/etc/init.d/network'
Jan  1 12:00:01 (none) daemon.info init: starting pid 719, tty '': '/etc/init.d/chronyd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 720, tty '': '/etc/init.d/nfsd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 721, tty ' [... cut]