Jalle19/satip-axe #72: Tuning fails with tvheadend

sakos, 2023-06-19:
Thank you very much for answering my previous noob questions. It helped a lot, but I am still struggling with setup Digibit R1 and Tvheaend server. When I try to start watching a channel I got tunning fails message in tvheadend. It is very odd that the it worked one day and does not work on the other day without changing the configuration.
One port of a quad LNB is connected to the first port of Digibit. The signal is OK, tv reception fine on the other output.

Nevertheless, this is the log of failing case from Tvheadend:
```
2023-06-19 19:59:56.720 subscription: C210: "192.168.10.1 [ taveleres | Kodi Media Center ]" subscribing on channel "RTL KETTO HD", weight: 150, adapter: "SAT>IP DVB-S Tuner #1 (192.168.10.44@TCP)", network: "Thor 0.8W", mux: "11804.2V", provider: "M7", service: "RTL KETTO HD", profile="htsp", hostname="192.168.10.1", username="taveleres", client="Kodi Media Center"
2023-06-19 19:59:56.737 satip: SAT>IP DVB-S Tuner #1 (192.168.10.44@TCP) - RTSP cmd error 7 (DONE) [8-404]
2023-06-19 19:59:56.777 satip: SAT>IP DVB-S Tuner #1 (192.168.10.44@TCP) - RTSP cmd error 7 (DONE) [8-404]
2023-06-19 19:59:58.719 subscription: C210: service instance is bad, reason: Tuning failed
2
```

and from Digibit (2 hours shift in time due to missing timezone settings):

```
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: sockets_add: handle 9 (type 1) returning socket sock 6 [192.168.10.42:52728] read: 0x4086b0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: minisatip.c:1047 get_sid returns NULL for s_id = -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: read RTSP (from handle 9 sock 6, len: 172, sid -1):
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: detect_dvb_parameters (S)-> src=1&fe=1&freq=11804.2&sr=30000&msys=dvbs2&mtype=8psk&pol=v&fec=34&ro=0.35&pids=0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: detect_dvb_parameters (E) -> src=1, fe=1, freq=11804200, fec=3, sr=30000000, pol=1, ro=0, msys=6, mtype=9, plts=2, bw=8000000, inv=2, pids=0 - apids=NULL - dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Setup stream -1 parameters, sock_id 6, handle 9
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: stream.c:197 get_sid returns NULL for s_id = -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: set_sock_lock: sock_id 6 locks also mutex 0x4b0a0c
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: sockets_add: handle -2 (type 0) returning socket sock 7 [0.0.0.0:0] read: 0x40872c
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: set_sock_lock: sock_id 7 locks also mutex 0x4b0a0c
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Setup stream done: sid: 0 for sock 6 handle 9
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: copy_dvb_param start -> src=0, fe=0, freq=0, fec=9, sr=0, pol=0, ro=3, msys=0, mtype=6, plts=2, bw=8000000, inv=2, pids=NULL, apids=NULL, dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: copy_dvb_parameters -> src=1, fe=1, freq=11804200, fec=3 sr=30000000, pol=1, ro=0, msys=6, mtype=9, plts=2, bw=8000000, inv=2, pids=0, apids=NULL, dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Assuming RTSP over TCP for stream 0, arg RTP/AVP/TCP;interleaved=0-1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: output socket buffer size for socket 9 is 1048576 bytes
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: reply -> 9 (192.168.10.42:52728) CL:0 [sock_id 6]:
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: read RTSP (from handle 9 sock 6, len: 160, sid 0):
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: detect_dvb_parameters (S)-> src=1&fe=1&freq=11804.2&sr=30000&msys=dvbs2&mtype=8psk&pol=v&fec=34&ro=0.35&pids=0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: detect_dvb_parameters (E) -> src=1, fe=1, freq=11804200, fec=3, sr=30000000, pol=1, ro=0, msys=6, mtype=9, plts=2, bw=8000000, inv=2, pids=0 - apids=NULL - dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Setup stream 0 parameters, sock_id 6, handle 9
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: copy_dvb_param start -> src=1, fe=1, freq=11804200, fec=3, sr=30000000, pol=1, ro=0, msys=6, mtype=9, plts=2, bw=8000000, inv=2, pids=0, apids=NULL, dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: copy_dvb_parameters -> src=1, fe=1, freq=11804200, fec=3 sr=30000000, pol=1, ro=0, msys=6, mtype=9, plts=2, bw=8000000, inv=2, pids=0, apids=NULL, dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Play for stream 0, type 3, rsock 9, adapter -1, sock_id 6, rsock_id 6, handle 9
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: stream.c:267: get_adapter returns NULL for adapter_id -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: adapter.c:1183: get_adapter returns NULL for adapter_id -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: adapter.c:2222: get_adapter returns NULL for adapter_id -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: stream.c:276: get_adapter returns NULL for adapter_id -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: get free adapter 1 - a[0] => e:0 m:-1 sid_cnt:0 f:11804200 pol=1 sys: dvbs undefined
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: trying to open [0] adapter 0 and frontend 0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: opened DVB adapter 0 fe:10 dvr:11
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: deleting pids on adapter 0, sid -1, pids=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: sockets_add: handle 11 (type 5) returning socket sock 8 [0.0.0.0:0] read: 0x4086b0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: set_socket_thread: thread 296c8000 for sockets 8
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: done opening adapter 0 delivery systems: dvbs undefined undefined undefined
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Got adapter 0 on stream 0 socket 6
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: set adapter 0 for stream 0 m:0 s:1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: setting DVB parameters for adapter 0 - master_sid 0 sid 0 old f:0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: new parameters: f:11804200, plp/isi:-1, diseqc:1, pol:1, sr:30000000, mtype:9
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: old parameters: f:0, plp/isi:-1, diseqc:0, pol:0, sr:0, mtype:6
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: deleting pids on adapter 0, sid -1, pids=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: copy_dvb_param start -> src=0, fe=0, freq=0, fec=9, sr=0, pol=0, ro=3, msys=0, mtype=6, plts=2, bw=8000000, inv=2, pids=NULL, apids=NULL, dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: copy_dvb_parameters -> src=1, fe=1, freq=11804200, fec=3 sr=30000000, pol=1, ro=0, msys=6, mtype=9, plts=2, bw=8000000, inv=2, pids=0, apids=NULL, dpids=NULL x_pmt=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: deleting pids on adapter 0, sid 0, pids=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: adding pids to adapter 0, sid 0, pids=0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Dumping pids table for adapter 0, pid errors 0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: pid 0, fd 0, packets 0, d/c errs 0/0, flags 2, pmt -1, filter -1, sock -1, sids: 0 -1 -1 -1 -1 -1 -1 -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: unable to find input for diseqc 0 (absolute switch), adapter 0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Frequency 0 is not within range
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: deleting pids on adapter 0, sid 0, pids=NULL
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Dumping pids table for adapter 0, pid errors 0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: pid 0, fd 0, packets 0, d/c errs 0/0, flags 3, pmt -1, filter -1, sock -1, sids: -1 -1 -1 -1 -1 -1 -1 -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Dumping pids table for adapter 0, pid errors 0
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: pid 0, fd 0, packets 0, d/c errs 0/0, flags 3, pmt -1, filter -1, sock -1, sids: -1 -1 -1 -1 -1 -1 -1 -1
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: reply -> 9 (192.168.10.42:52728) CL:0 [sock_id 6]:
Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: read RTSP (from handle 9 sock 6, len: 201, sid 0):
[... 59 lines cut ...]
[... cut ...]
Jun 19 18:00:01 (none) daemon.notice minisatip[1035]: get_signal took 15 ms for adapter 0 handle 10 (status: 32, ber: 0, strength:0, snr: 255, force scan 0)
Jun 19 18:00:02 (none) daemon.notice minisatip[1035]: get_signal adapter 0: status 32, strength 0, snr 65535, BER: 0
Jun 19 18:00:02 (none) daemon.notice minisatip[1035]: get_signal took 15 ms for adapter 0 handle 10 (status: 32, ber: 0, strength:0, snr: 255, force scan 0)

```
with Wireshark I saw  RTSP/1.0 404 Not Found response form the Digibit

Can you please guide me what can be the root cause of the issue?

Jalle19, 2023-06-20:
> Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: unable to find input for diseqc 0 (absolute switch), adapter 0
> Jun 19 17:59:56 (none) daemon.notice minisatip[1035]: Frequency 0 is not within range

This looks suspicious. What options do you run minisatip with?

sakos, 2023-06-20:
MINISATIP_OPTS="-e 0-3 -d 0:0-0,1:0-0,2:1-0,3:0-1 -A 1:0:0,1:1:0,1:2:1,1:3:1,3:2:2,3:3:3,4:3:4,5:3:5,6:2:3,6:3:6,7:3:7,8:3:8"

Port1, port2: direct LNB on 0,8W (no diseqc switch)
Port3: 4port switch 
Port4: 10 port switch

Jalle19, 2023-06-20:
Try unchecking "FE supported" in tvheadend, on the tuner page

sakos, 2023-06-20:
I started to going crazy after spending a couple of hours to find some systematic behavior. Finally made a step ahead. 
Setting FE supported on or off did not make any change. I went back to a single LNB on a single port setup and found that my -A parameters are wrong. -A 1:0:0... is not working, -A **0**:0:0... works. It is strange because an input on the 10 port switch seemed working before with the settings above.
I wonder how the correct SRC numbers shall be correctly set.

The issue has another part. The succes of satip tuning depends on the availability of conventional tuners. If there is no other tuner, satip works. If there is a free conventional tuner then satip tuning fails. Very odd. I assume that the tvheadend server gives up the  tuning too fast and select next tuner if there is a free available. I am going to investigate this more deeply and get some logs.

Jalle19, 2023-06-21:
I would disable all physical tuners in tvheadend so you can isolate the issue

sakos, 2023-06-25:
Hi,

The tuning problem is solved. The virtual-diseqc mapping had to be completely removed. I misunderstood the aim of this parameter. The satellite positions was configured in tvheadend, hence no further mapping was needed.
The topic of the ticket is solved, hence it can be closed.

Nevertheless, I am still not happy. Hopefully you can guide me.
There are sporadic disturbances, mainly on HDTV channels. Around the disturbances the stats on webgui of minisatip shows sudden drop in bandwith ~1500KB/s to ~250KB/s, sometimes dropped and buffered counters appear.
Signal quality SNR:~96-99%, STR:~82%.
TVH server is running on an RPi3 with latest Librelec and addon versions.

Absolutely no disturbance is visible on the same channel with legacy tuner at the same time.
It is very bad, because my expectation was a more robust system than my old outdated Wetek Play box.

Jalle19, 2023-06-25:
Try ticking "RTP/TCP supported" in tvheadend on the settings page for your minisatip server.

sakos, 2023-06-25:
It is already set.

Jalle19, 2023-06-25:
I would try streaming from another device than your Raspberry Pi, just to rule out any issues with it. Your description makes it sound like slow wifi or just otherwise a flaky network.

sakos, 2023-06-25:
Without the RTP/AVP/TCP is set, there are continous picture disturbances. I assume that on a wired LAN connection even pure UDP should work flawless. Forcing TCP is a kind of dirty workaround.
Thanks for your suggestion. I will try with another hardware for the TVH server.

Anyway, can you confirm that handling (1-4?) HDTV channels shall not cause performance problems on the Digibit?

sakos, 2023-06-25:
New finding.
By setting the switch port to 100Mbps half-duplex, the picture is good even with no TCP configured. Something is fishy with the LAN connection between the digibit and the switch. It seems not only the 1G bit speed is affected.

closed 2023-06-25 by sakos

Jalle19, 2023-06-25:
> Anyway, can you confirm that handling (1-4?) HDTV channels shall not cause performance problems on the Digibit?

Yes, it should absolutely not be a problem. I haven't found any upper limit on my two devices, you can use all tuners at once and multiple channels per tuner.