Jalle19/satip-axe #75: wget: can't execute 'ssl_helper': No such file or directory

pbriesch, 2023-06-29:
Just wanted to upgrade from ..-rc1.fw, but I cannot download the firmware. 

Is `ssl_helper` missing?

```
~ # wget https://github.com/Jalle19/satip-axe/releases/download/build25-rc2/sati
p-axe-202306291345-25-rc2.fw
Connecting to github.com (140.82.121.3:443)
wget: can't execute 'ssl_helper': No such file or directory
wget: error getting response: Connection reset by peer
```

closed 2023-06-30 by Jalle19

Jalle19, 2023-06-30:
https://github.com/Jalle19/satip-axe/releases/tag/build25-rc3

@pbriesch you'll have to upload the firmware file with `scp` if you want to upgrade this time.

MusterGit, 2023-06-30:
scp doesn't work for me since rc1. I'm getting this error:
```
/usr/libexec/sftp-server: error while loading shared libraries: libdl.so.2: cannot open shared object file: No such file or directory
scp: Connection closed
```

pbriesch, 2023-06-30:
@MusterGit 

I had this problem with scp as well. Here I found the solution: [https://forum.openwrt.org/t/ash-usr-libexec-sftp-server-not-found-when-using-scp/125772/3](url)

In short, use the option `-O`, capital letter O as in Oscar.  `scp -O`

Jalle19, 2023-06-30:
Well, shit :grin:

Jalle19, 2023-06-30:
https://github.com/Jalle19/satip-axe/pull/78

Jalle19, 2023-06-30:
Turns out the same fix that fixed `wget` also fixed the SFTP issue

PR #76 (2023-06-30): Copy the "openssl" binary, turns out wget needs it
Fixes #75
commit 5801b266 (2023-06-30): Copy the "openssl" binary, turns out wget needs it

Fixes #75