Jalle19/satip-axe #79: Support satip-axe  Digibit R1 for DVB-S2 with PLS?

roisiano, 2023-09-07:
Hello:

I am not sure whether it is a issue or not, but I'm trying to tune on Hispasat 30W:
https://es.kingofsat.net/pos-30W.php 

Transponders:

* 11330.00 | H | 128 | Europe | DVB-S2PLS: Gold+174526 | 8PSK | 9140 3/4

* 1491.00 | V | 199 | Europe | DVB-S2PLS: Gold+79463 | 8PSK | 7522 8/9

tvheadend, TP created:
https://images2.imgbox.com/e6/19/w1UoaAnu_o.png

Well: when I try to tune it (using SATIP-AXE server 202212192117-24-rc3):
https://images2.imgbox.com/a0/8e/aUfhqj0B_o.png 

No signal (SNT). No service found.

I tune in without problems the rest of the "normal" transponders of Hispasat, all services found, etc., but NOT from these 2 "particular" TP.

Could it be a limitation of Digibit R1? SATIP-AXE issue?
These 2 transponders have too low signal? 

I have more questions than answers, that is the truth. 

Kind regards

Jalle19, 2023-09-08:
I think this is a minisatip issue, you might have better luck opening an issue there (https://github.com/catalinii/minisatip)

roisiano, 2023-10-03:
Hello:

Right now I just tuned the channels of:
11330.00H, 8PSK, 9140, 3/4 (DVB-S2 ; PLS: Gold+174526).

https://es.kingofsat.net/pos-30W.php

I switched to the latest firmware ( https://github.com/Jalle19/satip-axe/releases/tag/build25 ) this weekend, so I'm not sure if it was a software-firmware problem or weak signal issue ( I receive the TP with a very low signal). I would bet on a signal problem.

SOLVED!

Jalle19, 2023-10-03:
Glad you got it working, I think we should close this and open new tickets if any more specific issues arise.

closed 2023-10-03 by Jalle19