Jalle19/satip-axe #81: Software watchdog

melias122, 2023-11-29:
Sometimes my Digibit R1 is stucked and it cannot be reached on network (ping) and I need to restart it manually. 

It would be nice to have a software [watchdog](https://manpages.ubuntu.com/manpages/jammy/en/man8/watchdog.8.html), where I could specify e.g. ping ip address and it would restart if not reachable.

This is up to discussion, I am not sure if its even possible on this machine/old linux.

roisiano, 2023-12-05:
> I could specify e.g. ping ip address and it would restart if not reachable.
> 
> This is up to discussion, I am not sure if its even possible on this machine/old linux.

I'm interested, too.

I live a little far from where I have the Digibit R1 working, so my (physical)  "watchdog" is my father (xD).

So, sometimes, I have to use the telephone and...
"Please, dad, turn the Digibit off and on again..."

Jalle19, 2023-12-05:
I'm not sure if a watchdog process would be able to catch the kind of failures that require someone to physically reset the device. If anyone has a suggestion on what settings to use for the watchdog I'd be willing to try it.

melias122, 2023-12-08:
I believe it would. But I also do not have experience with it..

Currently I am using cron+script and it works just fine.
```
$ cat /etc/sysconfig/crontabs/root 
* * * * * /etc/sysconfig/watchdg
```

Trying to reach tvheadend server
```
$ cat /etc/sysconfig/watchdg
#/bin/sh
echo -e "GET / HTTP/1.0\n\n" | nc -w 30 192.168.1.45 9981 > /dev/null 2>&1
if [ $? = 1 ]; then
    reboot
fi
```

Jalle19, 2024-04-12:
There's a built-in watchdog daemon, perhaps we could use it?

```
~ # watchdog --help
BusyBox v1.26.2 (2023-08-06 10:58:40 UTC) multi-call binary.

Usage: watchdog [-t N[ms]] [-T N[ms]] [-F] DEV

Periodically write to watchdog device DEV

	-T N	Reboot after N seconds if not reset (default 60)
	-t N	Reset every N seconds (default 30)
	-F	Run in foreground

Use 500ms to specify period in milliseconds
```

I don't want to add any functionality that depends on certain network targets being reachable, since the checks may fail even though there's nothing wrong with the Digibit device itself.

closed 2025-11-29 by Jalle19