Jalle19/satip-axe #84: Tuner disabled

VenimK, 2024-04-20:
Had a powersurge.

Got my dgibit R1 back up and running
but my tuners keep getting 'disabled'

> Linux version 2.6.32.42_stm24_0208-idl4k_7108 (root@5e9872285089) (gcc version 4.8.4 20141219 (STMicroelectronics/Linux Base 4.8.4-139) (GCC) ) #1 PREEMPT Sun Aug 6 11:05:32 UTC 2023
Boot params:
... MOUNT_ROOT_RDONLY - 00000001
... RAMDISK_FLAGS     - 00000000
... ORIG_ROOT_DEV     - 00000200
... LOADER_TYPE       - 00000000
... INITRD_START      - 00000000
... INITRD_SIZE       - 00000000
Booting machvec: idl4k
Node 0: start_pfn = 0x70000, low = 0x8f000
Zone PFN ranges:
  Normal   0x00070000 -> 0x0008f000
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0: 0x00070000 -> 0x0008f000
On node 0 totalpages: 126976
free_area_init_node: node 0, pgdat 813fd7dc, node_mem_map 82ac3000
  Normal zone: 992 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 125984 pages, LIFO batch:31
Inverto idl4k board initialisation
STx7108 version 2.x, HOST core
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line:  console= hw=16001000000000000000000000000000 sn=1S81T51717001207 uuid=6BF3D2EE4EBF4B0E9CB6DE7C97021AFA revBoot=stm24_0056-idl4k console=ttyAS0,115200 bigphysarea=20000
bpa2: partition 'bigphysarea' created at 0x72ea7000, size 80000 kB (0x04e20000 B)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
PVR=04909300 CVR=60880000 PRR=0000a320
I-cache : n_ways=2 n_sets=512 way_incr=16384
I-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
D-cache : n_ways=2 n_sets=512 way_incr=16384
D-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
Memory: 379520k/507904k available (2998k kernel code, 1088k data, 22188k init)
Hierarchical RCU implementation.
NR_IRQS:500
Failed to enable clk CLKA0_REF, ignoring
Failed to enable clk CLKA0_PLL0LS, ignoring
Failed to enable clk CLKA1_REF, ignoring
Failed to enable clk CLKA1_PLL0LS, ignoring
Console: colour dummy device 80x25
console [ttyAS0] enabled
sh_tmu: TMU0 used for clock events
sh_tmu: TMU0 used for periodic clock events
sh_tmu: TMU1 used as clock source
Calibrating delay loop... 595.96 BogoMIPS (lpj=297984)
Mount-cache hash table entries: 512
CPU: STx7108
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource TMU1
NET: Registered protocol family 2
IP route cache hash table entries: 4096 (order: 2, 16384 bytes)
TCP established hash table entries: 16384 (order: 5, 131072 bytes)
TCP bind hash table entries: 16384 (order: 4, 65536 bytes)
TCP: Hash tables configured (established 16384 bind 16384)
TCP reno registered
NET: Registered protocol family 1
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
[... 228 lines cut ...]
DROPBEAR="yes"

#
# minisatip
#
MINISATIP="yes"
MINISATIP_OPTS="-f --diseqc-timing *:155-54-15-15-15-0 -7 0:1,0:2,0:3"

#
# inetd (httpd,ftpd)
#
INETD="yes"

#
# crond
#
CROND="yes"

#
# oscamd
#
OSCAMD="yes"

Any Problems here
or any tips what I can check

Jalle19, 2024-04-20:
What do you mean by "disabled"?

VenimK, 2024-04-21:
[image]
[image]

Jalle19, 2024-04-21:
That is normal, the tuners will wake up when you try to use them.

Jalle19, 2024-06-14:
Closing this, not clear if there is any actual issue

closed 2024-06-14 by Jalle19