Jalle19/satip-axe #91: minisatip not work

Marvicek, 2024-11-01:
Hello
I have gg sat-ip i used revision  15 for 6 years. But now i lost sat- ip tuner in tvheadend. I must restart box manualy and after this work  for 5 days. I try last revision of fimware from you. 26 but i am able to connect via putty.. Oscam webinterface working but i am not able to see 8080 port of sat ip. also tvheadend do not see tunners. 

If i go back to the old firmware .. working with restart. .

What is problem ? 
minisat ip not start?

Marvicek, 2024-11-01:
Hello i try revision 16 from mr kysela --- link to you repository .. and working... your not.

Jalle19, 2024-11-01:
Can you post the contents of /etc/sysconfig/config?

Jalle19, 2024-11-14:
You will most likely have to update your configuration so it uses these two options:

```
MINISATIP="yes"
MINISATIP_OPTS=""
```

closed 2024-11-14 by Jalle19