Jalle19/satip-axe #92: minisatip isnt disoverable in tvheadend

PANiCnz, 2024-11-09:
Has anyone else had a problem where the minisatip server isn't discoverable in tvheadend? 

The R1 successfully boots and the OSCAM web interface is accessible.

Jalle19, 2024-11-10:
Should work just fine. Is it on the same network as tvheadend?

PANiCnz, 2024-11-11:
Same network. Not change except moving from Build 15 to Build 26. Remove the USB key with Build 26 on it, reverted back to Build 15 and it was detected within a few minutes.

Jalle19, 2024-11-11:
The log that generates the UUID has changed, so when you upgrade to Build 26 it won't show up as the same server but as a new one. Are you sure it doesn't show up at all? Can you post the command-line used to run minisatip (can be seen in the footer of the web interface)?

Jalle19, 2024-11-14:
Also if you've upgraded from a very old build you will have to update your configuration file to match, otherwise minisatip will most likely run with default settings only. See https://github.com/Jalle19/satip-axe/issues/91.

Jalle19, 2024-11-14:
Please post your configuration

PANiCnz, 2024-11-16:
Configuration is stock. Does it need to be updated?

Jalle19, 2024-11-17:
Yes, but since I don't remember what the stock configuration from 7 years ago looks like I need you to post it.

Jalle19, 2024-11-29:
Closing until there's more activity

closed 2024-11-29 by Jalle19

PANiCnz, 2024-11-30:
This is my current config that works with build 15 and 16, I cant see any obvious difference between the config file for build 26. From what I can see minisatip isn't starting on boot, the web interface on 8080 isn't accessible.

`#
# SATIP AXE firmware configuration
#

CFGVER="14"
HOSTNAME="satip-axe"
KMODULES="drivers/usb/serial/cp210x.ko drivers/usb/serial/pl2303.ko drivers/usb/serial/ftdi_sio.ko"

#
# STV6210 baseband amplifier gain (0dB .. 16dB - 2dB step)
#
STV6210_BAMP="16dB"
#DEMUXTS_OLD="yes"

#
# ttyUSB names
#

TTYUSB1_DEV="2-1:1.0"
TTYUSB1_NAME="ttyUSBupper"
TTYUSB2_DEV="4-1:1.0"
TTYUSB2_NAME="ttyUSBlower"

TTYUSB3_DEV="1-1.1:1.0"         # USB hub port#1 in upper port
TTYUSB3_NAME="ttyUSBupper1"
TTYUSB4_DEV="3-1.2:1.0"         # USB hub port#2 in lower port
TTYUSB4_NAME="ttyUSBlower2"

#
# IPV4 config
#
IPV4="dhcp"
# or
#IPV4="192.168.1.240"
#IPV4_NETMASK="255.255.255.0"
#IPV4_BROADCAST="192.168.1.255"
#IPV4_GATEWAY="192.168.1.1"

#
# ntpd
#
NTPD_OPTS="-p 0.pool.ntp.org"

#
# syslogd
#
#SYSLOGD_OPTS="-R 192.168.1.1:514"

#
# telnetd
#
TELNETD="yes"

#
# dropbear
#
DROPBEAR="yes"

#
# minisatip
#
#MINISATIP="yes"
#MINISATIP_OPTS=""

#
# minisatip 0.5
#
#MINISATIP5="yes"
#MINISATIP5_OPTS=""

#
# minisatip 0.7
#
#MINISATIP7="yes"
#MINISATIP7_OPTS=""

#
# minisatip 0.8
#
MINISATIP8="yes"
MINISATIP8_OPTS="-e 0-3 -7 0:1,0:2,0:3 -L *:10750-10750-99999"

#
# inetd (httpd,ftpd)
#
INETD="yes"

#
# crond
#
CROND="yes"

#
# nfsd v3
#
NFSD="yes"
NFSD_HOTPLUG_EXPORT="*" # replace with host or subnet, '*' = world

#
# oscamd
#
OSCAMD="yes"

#
# custom daemon/service/setup executed from sysinit
#
#CUSTOMD="/etc/sysconfig/myboot"
`

Jalle19, 2024-11-30:
The problem is that your old configuration uses `MINISATIP8` while the new one only uses `MINISATIP`. I suggest you start with a fresh configuration based on https://github.com/Jalle19/satip-axe/blob/99781bd8d73d1a5fafacbc32f15d895ae5bc9a86/fs-add/etc/config.default, apply your minisatip configuration in `MINISATIP_OPTS`, then reboot.

Jalle19, 2024-11-30:
https://github.com/Jalle19/satip-axe?tab=readme-ov-file#updating-from-older-upstream-builds-16-and-older I added a note to the README regarding this

PANiCnz, 2024-11-30:
Thanks that sorted it. I think README can still be clearer. 

I wasn't using the config file from build 16 I was using the stock out of the box config file for 26. There is nothing in the documentation that indicates MINISATIP8 needs to be disabled and MINISATIP enabled. 

Now that it's up and running I'm having issues with the second tuner, it's got poor signal strength even though the device is configured to share the input from LNB1. Any advice on config that might improve the situation? 

[image]

Jalle19, 2024-12-01:
If you upgraded from an old build your old configuration will still be in use. The stock configuration is only applied if you remove /etc/sysconfig/config.