Jalle19/satip-axe #93: kernel-module i2c_mangle blocks all DVBS2 channels

ast261, 2024-12-06:
Starting with build 25 and 26 all(!) DVBS2 channels show SNR 0% (instead of 100%) and tvheadend says:
> service instance is bad, reason: No input detected

Reason for this is the newly added kernel-module  `i2c_mangle`.
When I do not insert this module, everything is fine.

Jalle19, 2024-12-06:
That's strange, what hardware do you use?

Jalle19, 2024-12-06:
Works fine for me. Wonder what makes it fail in your case.

ast261, 2024-12-10:
Can u please attach a kernel startup log?

Jalle19, 2024-12-10:
```
~ # dmesg
Linux version 2.6.32.42_stm24_0208-idl4k_7108 (@333691d388cc) (gcc version 4.8.4 20141219 (STMicroelectronics/Linux Base 4.8.4-139) (GCC) ) #1 PREEMPT Tue Sep 17 09:34:34 UTC 2024
Boot params:
... MOUNT_ROOT_RDONLY - 00000001
... RAMDISK_FLAGS     - 00000000
... ORIG_ROOT_DEV     - 00000200
... LOADER_TYPE       - 00000000
... INITRD_START      - 00000000
... INITRD_SIZE       - 00000000
Booting machvec: idl4k
Node 0: start_pfn = 0x70000, low = 0x8f000
Zone PFN ranges:
  Normal   0x00070000 -> 0x0008f000
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0: 0x00070000 -> 0x0008f000
On node 0 totalpages: 126976
free_area_init_node: node 0, pgdat 813fd7dc, node_mem_map 82aef000
  Normal zone: 992 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 125984 pages, LIFO batch:31
Inverto idl4k board initialisation
STx7108 version 2.x, HOST core
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line:  console= hw=16001000000000000000000000000000 sn=1S81T51717003193 uuid=D080FC64220449CE93666F00EED4C465 revBoot=stm24_0056-idl4k console=ttyAS0,115200 bigphysarea=20000
bpa2: partition 'bigphysarea' created at 0x72ed3000, size 80000 kB (0x04e20000 B)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
PVR=04909300 CVR=60880000 PRR=0000a320
I-cache : n_ways=2 n_sets=512 way_incr=16384
I-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
D-cache : n_ways=2 n_sets=512 way_incr=16384
D-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
Memory: 379392k/507904k available (2998k kernel code, 1088k data, 22364k init)
Hierarchical RCU implementation.
NR_IRQS:500
Failed to enable clk CLKA0_REF, ignoring
Failed to enable clk CLKA0_PLL0LS, ignoring
Failed to enable clk CLKA1_REF, ignoring
Failed to enable clk CLKA1_PLL0LS, ignoring
Console: colour dummy device 80x25
console [ttyAS0] enabled
sh_tmu: TMU0 used for clock events
sh_tmu: TMU0 used for periodic clock events
sh_tmu: TMU1 used as clock source
Calibrating delay loop... 595.96 BogoMIPS (lpj=297984)
Mount-cache hash table entries: 512
CPU: STx7108
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource TMU1
NET: Registered protocol family 2
IP route cache hash table entries: 4096 (order: 2, 16384 bytes)
TCP established hash table entries: 16384 (order: 5, 131072 bytes)
TCP bind hash table entries: 16384 (order: 4, 65536 bytes)
TCP: Hash tables configured (established 16384 bind 16384)
TCP reno registered
NET: Registered protocol family 1
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
[STM]: [PM]: HoM support registered
[STM]: [PM]: [HoM]: Early console @ 0xfd733000
Slow work thread pool: Starting up
[... 148 lines cut ...]
[fta_fe] FE Initialization done, res 0
[dmx-ts] ... init 0
USB Serial support registered for cp210x
usbcore: registered new interface driver cp210x
cp210x: v0.09:Silicon Labs CP210x RS232 serial adaptor driver
USB Serial support registered for pl2303
usbcore: registered new interface driver pl2303
pl2303: Prolific PL2303 USB to serial adaptor driver
USB Serial support registered for FTDI USB Serial Device
usbcore: registered new interface driver ftdi_sio
ftdi_sio: v1.5.0:USB FTDI Serial Converters Driver
eth0: device MAC address 00:24:a6:00:2e:87
STMMAC MII Bus: Can't get IRQ 25 (PHY)
 No MAC Management Counters available
PHY: 1:01 - Link is Up - 1000/Full
eth0: no IPv6 routers present
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
```

Jalle19, 2024-12-10:
@ast261 was added here: https://github.com/Jalle19/satip-axe/pull/67, there's some explanation of what it does

ast261, 2024-12-19:
The kernel log is identical to mine.

I tested all STV6210 baseband amplifier gain settings from 0dB to 16dB.
All with the same result.

Jalle19, 2024-12-19:
Strange

perexg, 2024-12-19:
The gain settings (which i2c_mangle sets) is in config file:

```
#
# STV6210 baseband amplifier gain (0dB .. 16dB - 2dB step)
#
STV6210_BAMP="16dB"
#DEMUXTS_OLD="yes"
```

The 16dB may not be appropriate for all inputs. Try to lower this value. Perhaps, your input level is already too strong.

ast261, 2024-12-23:
I have tried all values described in the axe-settings script.

My setup is the following:

I have a quattro switch LNB.
But the digibit is only connected to one LNB.

Jalle19, 2024-12-23:
Maybe we should just disable the module by default? I'm not sure if it brings any benefit to the average user.

ast261, 2024-12-24:
Your decision.
I am able to build an image on my own.

As I am currently the only user with that problem, I would wait.

Jalle19, 2024-12-24:
Sounds reasonable. Let's leave this ticket open and see if anyone reports any issues.