Jalle19/satip-axe #97: No Tuners Found ....

VenimK, 2025-02-28:
[demod900] setup [num_demods 4]
[pll6120] setup [num_plls:4] [clk:30000000] [bbgain:16]
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD0682
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD06B0
[i2c] I2C bus error /errno -121/!!
[demod900-0] i2c bus error, unable to read ChipID
[i2c] I2C bus error /errno -121/!!
[i2c] I2C bus error /errno -121/!!
[i2c] I2C bus error /errno -121/!!
[i2c] I2C bus error /errno -121/!!
[i2c] I2C bus error /errno -121/!!
[i2c] I2C bus error /errno -121/!!
[pll6120] init error!
[fta_fe] FE Initialization done, res -1

Jalle19, 2025-02-28:
That does not look right, should look something like this:

```
[demod900] setup [num_demods 4]
[pll6120] setup [num_plls:4] [clk:30000000] [bbgain:16]
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD0682
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD06B0
[dmx] ... init 0
[fta_fe] FE Initialization done, res 0
[dmx-ts] ... init 0
```

VenimK, 2025-02-28:
I saw there a chip with a coil loose ....

Jalle19, 2025-02-28:
Closing this as a hardware issue then. Sorry to hear about your problems, it's a good device when it works!

closed 2025-02-28 by Jalle19

VenimK, 2025-02-28:
Could resolder the chip :)

I2C-Bus AXE mangle module loaded
[demod900] setup [num_demods 4]
[pll6120] setup [num_plls:4] [clk:30000000] [bbgain:16]
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD0682
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD06B0
[dmx] ... init 0
[fta_fe] FE Initialization done, res 0
[dmx-ts] ... init 0

Hell Ya :)

Jalle19, 2025-02-28:
Yeah, those settings should work