Jalle19/satip-axe #100: read error when recovering original image

nrpetonr, 2025-06-01:
trying to recover original image, unfortunately not able to do so due to nand read error. any ide for this? 

```
idl4k> run updatefw
(Re)start USB...
USB:   scanning bus for devices... 2 USB Device(s) found
       scanning bus for storage devices... 1 Storage Device(s) found
reading idl4k.scr

234 bytes read
## Executing script at 84000000
Saving Environment to SPI Serial Flash...
.

Loading from NAND 128MiB 3,3V 8-bit, offset 0x420000
   Image Name:   Linux-2.6.32.42_stm24_0208-idl4k
   Image Type:   SuperH Linux Kernel Image (gzip compressed)
   Data Size:    20218094 Bytes =  19.3 MiB
   Load Address: 81000000
   Entry Point:  81001000
reading NAND page at offset 0x1563000 failed
** Read error
```

other methods, e.g. booting from usb and/or reflashing fw, end up with network not being accessible.

```
idl4k> dhcp
Using MAC Address 00:24:A6:00:2B:A9
STM-GMAC: Unable to find a known PHY!
STM-GMAC: info: PHY at address=0x00 has ID=0x001cc915
STM-GMAC: info: PHY at address=0x01 has ID=0x001cc915
STM-GMAC: ERROR: no PHY detected
```

nrpetonr, 2025-06-01:
no problem, u'r doing anyway great job. 
it is pitty, i am sure fw itself loads, just this network not getting ip is strange - last 5 line of the log. 

```
Terminal ready
Linux version 2.6.32.42_stm24_0208-idl4k_7108 (root@1458dabb56e0) (gcc version 4.8.4 20141219 (STMicroelectronics/Linux Base 4.8.4-139) (GCC) ) #1 PREEMPT Tue Sep 17 09:58:44 UTC 2024
Boot params:
... MOUNT_ROOT_RDONLY - 00000001
... RAMDISK_FLAGS     - 00000000
... ORIG_ROOT_DEV     - 00000200
... LOADER_TYPE       - 00000000
... INITRD_START      - 00000000
... INITRD_SIZE       - 00000000
Booting machvec: idl4k
Node 0: start_pfn = 0x70000, low = 0x8f000
Zone PFN ranges:
  Normal   0x00070000 -> 0x0008f000
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0: 0x00070000 -> 0x0008f000
Inverto idl4k board initialisation
STx7108 version 2.x, HOST core
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line: console= bigphysarea=20000 axecfg=reset hw=16001000000000000000000000000000 sn=1S81T51717002459 uuid=F2EB7457132A4214963114186828642E revBoot=stm24_0056-idl4k console=ttyAS0,115200 bigphysarea=20000
bpa2: partition 'bigphysarea' created at 0x72ed3000, size 80000 kB (0x04e20000 B)
bpa2: partition 'bigphysarea' created at 0x77cf4000, size 80000 kB (0x04e20000 B)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
PVR=04909300 CVR=60880000 PRR=0000a320
I-cache : n_ways=2 n_sets=512 way_incr=16384
I-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
D-cache : n_ways=2 n_sets=512 way_incr=16384
D-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
Memory: 299392k/507904k available (2998k kernel code, 1088k data, 22364k init)
Hierarchical RCU implementation.
NR_IRQS:500
Failed to enable clk CLKA0_REF, ignoring
Failed to enable clk CLKA0_PLL0LS, ignoring
Failed to enable clk CLKA1_REF, ignoring
Failed to enable clk CLKA1_PLL0LS, ignoring
Console: colour dummy device 80x25
console [ttyAS0] enabled
sh_tmu: TMU0 used for clock events
sh_tmu: TMU0 used for periodic clock events
sh_tmu: TMU1 used as clock source
Calibrating delay loop... 595.96 BogoMIPS (lpj=297984)
Mount-cache hash table entries: 512
CPU: STx7108
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource TMU1
NET: Registered protocol family 2
IP route cache hash table entries: 4096 (order: 2, 16384 bytes)
TCP established hash table entries: 16384 (order: 5, 131072 bytes)
TCP bind hash table entries: 16384 (order: 4, 65536 bytes)
TCP: Hash tables configured (established 16384 bind 16384)
TCP reno registered
NET: Registered protocol family 1
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
[STM]: [PM]: HoM support registered
[STM]: [PM]: [HoM]: Early console @ 0xfd733000
Slow work thread pool: Starting up
Slow work thread pool: Ready
[... 136 lines cut ...]
Load module stpti5_ioctl [236]		by insmod (pid 473)
I2C-Bus AXE mangle module loaded
[demod900] setup [num_demods 4]
[pll6120] setup [num_plls:4] [clk:30000000] [bbgain:16]
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD0682
[fe] Checking lut 0xC1AD064C
[fe] Checking lut 0xC1AD06B0
[dmx] ... init 0
[fta_fe] FE Initialization done, res 0
[dmx-ts] ... init 0
USB Serial support registered for cp210x
usbcore: registered new interface driver cp210x
cp210x: v0.09:Silicon Labs CP210x RS232 serial adaptor driver
USB Serial support registered for pl2303
usbcore: registered new interface driver pl2303
pl2303: Prolific PL2303 USB to serial adaptor driver
USB Serial support registered for FTDI USB Serial Device
usbcore: registered new interface driver ftdi_sio
ftdi_sio: v1.5.0:USB FTDI Serial Converters Driver
eth0: device MAC address 00:24:a6:00:2b:a9
STMMAC MII Bus: Can't get IRQ 25 (PHY)
 No MAC Management Counters available
PHY: 1:01 - Link is Up - 1000/Full
```

Jalle19, 2025-06-01:
Make sure you have `IPV4="dhcp"` in `/etc/sysconfig/config`

Jalle19, 2025-06-01:
I haven't noticed any such issues so I doubt it's a firmware issue.

nrpetonr, 2025-06-01:
Make sure you have IPV4="dhcp" in /etc/sysconfig/config   -  how to access if the device does not get ip? possiblke via  uart? i am not aware...

to which device is your Digibit connected?  - i tried via switch, i tried direct connection to the router, but no difference. the device not getting IP, actuallyt router does not even see it...

What does logread | grep udhcpc show when you have trouble getting an IP address?   -  how to do this? such command is not availabl ein uboot console

nrpetonr, 2025-06-01:
interesting hint, checked, but all capacitors are fine. actually uart reports it be alive   PHY: 1:01 - Link is Up - 1000/Full

it will rather be some U-Boot’s inability to map the detected PHY ID (0x001cc915) to the RTL8211E driver due to masking or configuration issues. just do not know how to get out of it :-)

closed 2025-06-01 by Jalle19